trezor/trezor-firmware #407: Fix python-shamir-mnemonic to have the same API as in Core

tsusanka, 2019-08-14:
Python-shamir-mnemonic's slip39 has different API than core/trezor/crypto/slip39.py. We should unify this and fix this: https://github.com/trezor/trezor-firmware/commit/1b666804c079b20e0e9681c7fff7c29b0954f98d#diff-bb0c35e93f0cbad363f855ebe7e33c5aR158

tsusanka, 2019-08-14:
@matejcik can I leave this one up to you?

matejcik, 2019-08-22:
i think @andrewkozlik was against touching the shamir-mnemonic API, because other projects are already using it?

tsusanka, 2019-08-22:
Hmm, I don't think there are any _other_ projects. But maybe we don't need to do this, I'm not sure.

prusnak, 2019-08-22:
I have a preference to fix that while we can.

andrewkozlik, 2019-09-11:
The main purpose of the reference implementation is to give a demonstration of how the spec is functionally meant to be implemented and to produce test vectors. It is not meant to demonstrate how the spec is implemented in Trezor. Anyone can check that for themselves by looking at the firmware codebase. Naturally the reference implementation and the Trezor implementation have different needs (error handling, security, persistence scenarios and other workflow scenarios in Trezor).
1. If these needs can be harmonized and some API changes are required to do that, then let's do it and bring the two together.
2. If there is something wrong with the API in the reference implementation then please specify what exactly it is and let's fix that. But if there isn't, then there is no reason to mess-up the API in the reference implementation just because the core has it differently for various (possibly good and possibly bad) reasons.

The main reason why I am hesitant to make fundamental modifications to the API is that there are already several other implementations of SLIP-0039 which have adapted this API into different programming languages and I feel it's terribly unfair to change it in the reference implementation at this point. What it means is that in one implementation a function like combine_mnemonics() will produce the **master secret** from the mnemonics and in another implementation it will produce the **encrypted master secret**. PLEASE, let's avoid that at all cost if we can!

Some examples of other projects:
https://github.com/BlockchainCommons/sss
https://github.com/Seclabs-htc/slip39
https://github.com/ilap/slip39-js
https://github.com/ilap/slip39-dart

What I think we should do in terms of point 1 is to split some of the high-level functions in the reference implementation into smaller ones. We still keep the high-level functions, but they will call these smaller functions in sequence. This doesn't break anything and merely broadens the API, which is what was needed in the core implementation. If necessary, we can then rename the functions in the core in accordance with these changes. The difficult part is making-up clear and simple names for these smaller functions.

matejcik, 2019-09-11:
briefly looking through these implementation, it seems that their APIs are not too similar to ours:

iLap is using an object model, with "generate mnemonics" as `Slip39.from`, and "combine mnemonics" as `Slip39.recoverSecret` (which btw conflicts with the spec's usage of "RecoverSecret")

Seclabs slip39 high-level functions are called `generate_mnemonics` and `combine_mnemonics`, and take MS + passphrase. This seems identical to the original python-shamir-mnemonic code, and we can easily keep these names.

sss is calling it `sss_generate_shares` and `sss_combine_shares`, which matches the spec's "GenerateShares" and (unnamed but implied) "CombineShares"

Presumably the important terms are "generate" and "combine". So in order of preference, I'd use the following names:

1. nice API: `generate` and `combine`, for use as `shamir_mnemonic.generate`, `shamir_mnemonic.combine`.
2. spec-like: `generate_shares` and `combine_shares`.
2. keep names `generate_mnemonics` and `combine_mnemonics`, because we don't actually *need* to break compatibility.

Whichever we choose, **neither** name should be used in core! Instead, we need two new names for "generate from data(EMS, id, ...)" and "recover EMS(mnemonics)"

matejcik, 2019-09-11:
update: sss is actually also calling it `generate_mnemonics` and `combine_mnemonics`, which is a word in favor of keeping these names.

for the new functions, I propose `encode_parameters` and `decode_parameters`. It's clear from the names that these are building blocks (what is a "parameter"?), plus their type signatures are user-unfriendly, unlike the main generate/combine.

matejcik, 2019-09-13:
@andrewkozlik I'd propose the following signatures:

`encrypt`: MS, passphrase, exponent, id ==> EMS
`decrypt`: EMS, passphrase, exponent, id ==> MS

optionally:
`encode_bytes`: bytes ==> words
`decode_bytes`: words ==> bytes

`encode_parameters`: EMS, id, exponent ==> words
`decode_parameters`: words ==> EMS, id, exponent

and then:
```python
def generate_shares(MS, passphrase, exponent=1):
    id = _get_random_id()
    EMS = encrypt(MS, passphrase, exponent, id)
    return encode_parameters(EMS, id, exponent)

def combine_shares(words, passphrase):
    EMS, id, exponent = decode_parameters(words)
    return decrypt(EMS, passphrase, exponent, id)
```

andrewkozlik, 2019-09-13:
In principle this looks good.
- I think the parameter "words" above should be called "mnemonics".
- I wish we could think of a better name for `encode_parameters()` and `decode_parameters()`, because until you posted the signatures, I wasn't entirely sure what these functions would be doing. The names don't give me any indication that they create/combine the shares. Maybe something like `split_ems()` and `recover_ems()`?
- Have you considered adding the Share class that we now have in core? I am thinking it might be more convenient if `encode_parameters()` and `decode_parameters()` worked with Share objects instead of mnemonics. During a previous code review I proposed that we move the functions `decode_mnemonic()` and `encode_mnemonic()` to the Share class and rename them to `from_mnemonic()` and `to_mnemonic()`. This would break the API here, but it would not change the meaning of any functions, so I am not against it if you think it's a good thing to do. It's better to do it now than later.

PR #750 (2020-01-09): core: add documentation for SLIP-39 handling in Trezor
Will fix #407 when done. This is now a draft of the documentation.
closed 2020-01-09 by matejcik