trezor/trezor-firmware #409: Use the same mnemonic dialog for BIP-39 as for SLIP-39

tsusanka, 2019-08-14:
The UI when showing mnemonic words is now different for BIP-39 and SLIP-39. Let's merge it into one (the SLIP-39 one).

BIP:
[image] [image]

SLIP:
[image] [image]

ciny, 2019-08-29:
While unifying the layouts I found that when confirming the displayed share words bip39 sends ButtonRequest.ResetDevice while slip39 sends ButtonRequest.Other. Unified to ResetDevice since it makes more sense in this context.

ciny, 2019-08-29:
Fixed, will be part of PR for #456

ciny, 2019-09-04:
implemented in #484

closed 2019-09-04 by ciny