trezor/trezor-firmware #410: Problem with signing big messages

bosomt, 2019-08-14:
If you try to sign big message it will get really slow and data overflows.

See screen and video

https://photos.app.goo.gl/zDAA2KNc7Bzs4qEz7

[image]

edit: 
fw 2.1.4, 
chrome and 
1.3.2 wallet

sorooris, 2019-08-21:
Please take into consideration that there's no option to scroll through the message on the device screen (T1 and/or TT) and therefore the user is not able to compare what is displayed in the wallet interface to what is on the device screen.

tsusanka, 2020-02-28:
@bosomt when you have a few seconds could you test this out? Not related to the current Freeze, use current master (5ee6e56). It is currently just a hotfix, where we shorten the text. We should introduce pagination later no to allow the user to read the whole message.

closed 2020-02-28 by tsusanka

bosomt, 2020-02-28:
@tsusanka working on it

bosomt, 2020-02-28:
2.1.1 vs 2.3.0
i don't see difference ;)

[image]
[image]

tsusanka, 2020-02-28:
It is different comparing to the first image.

bosomt, 2020-02-28:
BUT 2.1.4-2.3.0 is broken

so yes it it fixed. and sign speed is same compared to other versions

tsusanka, 2020-02-28:
Ok fixed then! (for now)

bosomt, 2020-03-23:
version:	2.3.0 RC1	revision:	6478b270
works fine, 
same speed and behavior with short or long message

sorooris, 2020-03-27:
2.3.0 8ae0535e
QA OK