trezor/trezor-firmware #416: [FW Feature Request] Signing support for arbitrary redeem/witness/scripts

brianddk, 2019-08-15:
With the [Bitcoin-HWI](https://github.com/bitcoin-core/HWI) support maturing, it would be nice to get support for arbitrary scripts.  P2SH and P2WSH have been fairly open ended in bitcoin-core for a while now.  Arbitrary script signing allows for features like CLTV as well as other exotic scripting solutions.

prusnak, 2019-08-16:
Please reopen when there is a BIP standard describing how this should work.

brianddk, 2019-08-16:
> Is there a BIP standard describing how this should work?

[BIP-16](https://github.com/bitcoin/bips/blob/master/bip-0016.mediawiki) (P2SH) lists multisig as the obvious use case for P2SH, but doesn't require that all redeem scripts be multisig.  In Bitcoin Core this interpretation was [formalized in v0.9.2](https://bitcoin.org/en/transactions-guide#pay-to-script-hash-p2sh).  Since [BIP-141](https://github.com/bitcoin/bips/blob/master/bip-0141.mediawiki#P2WSH) builds on previous P2SH implementations, the belief is that the ability to provide a generalized script as a redeem script or witness script (on P2WSH) is carried forward.  I can't find in the BIPs where this is formalized, but seems to be generally accepted (at least by the bitcoin core team).

[BIP-65](https://github.com/bitcoin/bips/blob/master/bip-0065.mediawiki) (CLTV) is the first time I see non-multisig scripts discussed in the BIPs, but seems to confirm the concept of general purpose scripting being acceptable if encapsulated in P2SH or P2WSH.  There are other examples of general purpose scripting in [Chapter 7 of Mastering Bitcoin](https://github.com/bitcoinbook/bitcoinbook/blob/develop/ch07.asciidoc#redeem-script-and-validation) though I wouldn't call these authoritative examples.

prusnak, 2019-08-17:
I think there is a lot of overlap with https://github.com/trezor/trezor-firmware/issues/49 (maybe even to the point this issue is a duplicate).

mattBlackDesign, 2019-12-13:
What is the progress on this? Ledger already has `signP2SHTransaction`

https://github.com/LedgerHQ/ledgerjs/tree/master/packages/hw-app-btc#signp2shtransaction

prusnak, 2019-12-16:
I agree it makes sense to do this, but we need to refactor the `wallet.sign_tx` app first: https://github.com/trezor/trezor-firmware/issues/617

esneider, 2020-06-20:
Now that #617 has been finished, are there any plans/roadmap for tackling this feature? Is there a way in which we might help it move faster?

It's a big blocker for most advanced vault-like use cases.

pavlovdog, 2020-10-22:
Hi, guys! Still in progress?

McLaughlinP, 2021-01-05:
I'm also curious about this feature - is this still in serious discussion?

prusnak, 2021-01-27:
It makes sense to combine this effort with the Taproot support in Trezor. Otherwise everything would need to be done twice (once for P2WPKH and once for P2TR).

MattG561, 2021-05-31:
Is this feature still being considered? It would drastically improve multi-signature flexibility.

landabaso, 2022-05-06:
Are there plans to support this?

As a side note, I see someone mentioned that ledger has signP2SHTransaction. Well, that method doesn't work with their redesigned hardware application either... At least they (still) embed the old HW App with the new one. But they plan to drop it.
The new HW app only signs txs that comply with accepted wallet policies. So far technically you can only sign legacy, segwit, nested segwit and multisitg with the new app. They plan to add support for other policies but, so far, the new app is more restrictive than the previous one. We'll see where this goes...

I'd like to encourage Trezor and Ledger to allow easier P2SH/P2WSH support. There are nice Bitcoin applications that could be developed that cannot make use of the Trezor hardware signature.

prusnak, 2022-05-06:
> Are there still plans to support this?

The main issue is not signing arbitrary scripts but how to describe the used scheme so Trezor can meaningfully show information about the signed data on the display. Signing blindly whatever script is being given is not secure. So far no one has came up with a reasonable way how to do it. There is one attempt here: https://lists.linuxfoundation.org/pipermail/bitcoin-dev/2022-May/020423.html The community has to agree first how to standardize the approach and  only after this is decided we can proceed with the implementation.

prusnak, 2022-05-06:
> And there is nothing that Trezor or Ledger can do to protect users who send funds to P2SH and P2WSH unclaimable addresses.

That's not true. You can provide a proof that you can spend the UTXOs. For each external key you provide proof of ownership, for internal keys you provide BIP32 path, so Trezor can derive they key and compare it.

landabaso, 2022-05-06:
> That's not true. You can provide a proof that you can spend the UTXOs. For each external key you provide proof of ownership, for internal keys you provide BIP32 path, so Trezor can derive they key and compare it.

What I meant is that hardware signing devices cannot prevent users from sending funds to burn addresses, unclaimable p2sh addresses, or hacker-controlled addresses, for example. If users are tricked to send funds to these addresses they will. There are things you can do to mitigate this, but it's impossible to protect all users from all risks. On the other hand, hardware signing devices may completely block users from spending their funded UTXOs. Anyway this was just a comment on how HD signing devices work, not a criticism.

> However, your comment nicely illustrates that the topic is not how to sign arbitrary script in a hardware wallet, but rather how to do all orchestration around it so it is still secure and understandable by the user signing.

Not an easy balance. Thanks for your comments.