trezor/trezor-firmware #426: master is broken because device tests do not recognize the new protobuf field

prusnak, 2019-08-21:
We've added the new field to protobuf messages here:

https://github.com/trezor/trezor-firmware/blob/4323b0ac2abb086afa95df6a1bbb4a788ecc476a/common/protob/messages-management.proto#L61-L77

Once I set the field here: https://github.com/trezor/trezor-firmware/blob/4323b0ac2abb086afa95df6a1bbb4a788ecc476a/legacy/firmware/fsm_msg_common.h#L86-L100

The device tests fail to parse the Features message and bail out.

prusnak, 2019-08-21:
IMO this should work, because I updated `python/src/trezorlib/messages/Feature{,s}.py` files and python-trezor is used from the currently check outed branch, not from the PyPI, right?

prusnak, 2019-08-21:
That said I am not sure python-trezor should bail out on new unrecognized fields at all. We do only forward-compatible changes, so we could afford skipping unrecognized fields IMO.

prusnak, 2019-08-21:
I removed the problematic code via https://github.com/trezor/trezor-firmware/commit/9d8ebfcdbc6118ade9583cce42a7aaf0b8cff4df so master should be OK again, but we need to fix this in order to merge `bitcoin-only` branch.

matejcik, 2019-08-22:
the problem is not in unrecognized fields but in unrecognized encoding (packed array)

it seems more difficult (or more hacky) to modify nanopb to not use packed arrays, so i'm implementing packed array support in trezorlib

commit 231d1504 (2019-08-22): python: implement decoding of protobuf packed repeated fields

also add typing
fixes #426
commit 132c8278 (2019-08-22): python: implement decoding of protobuf packed repeated fields

also add typing
fixes #426
closed 2019-08-22 by prusnak