trezor/trezor-firmware #437: Wanchain not signing anymore

tyrion70, 2019-08-22:
As discussed with @matejcik on Gitter. The derivation path should look at chain_id(=1 || 6) and txtype(=1) for Wanchain. Currently its sending a Forbidden key path error. 

Thanks!
Peter

prusnak, 2019-08-22:
Can you try and prepare a pull request?

tyrion70, 2019-08-22:
Sure! But can you give me a pointer where it's going wrong now? I see the error is thrown in core/src/apps/common/seed.py but tbh I can't really relate that to the def shortcut_by_chain_id that is already present in core/src/apps/ethereum/networks.py for Wan. It seems wan is not present in the .//common/defs/ethereum/networks.json either.
Could you give me some context as to where and why the error is thrown? Thx!

prusnak, 2019-08-22:
String search in the repo shows the following line: https://github.com/trezor/trezor-firmware/blob/0a594ea1e982a671f26cb448a5351b44443d2b0f/core/src/apps/common/seed.py#L62

tsusanka, 2019-08-23:
The paths are defined in the app's \_\_init\_\_ file see https://github.com/trezor/trezor-firmware/blob/0a594ea1e982a671f26cb448a5351b44443d2b0f/core/src/apps/ethereum/__init__.py#L13

matejcik, 2019-08-23:
i'm taking this, it's quicker to fix than it is to explain ;)

PR #440 (2019-08-25): core/ethereum: add support for wanchain slip44 id
fixes #437 

@tyrion70 please build this and ensure that it works fine with your wallet app
closed 2019-08-25 by prusnak