trezor/trezor-firmware #438: Cardano allows signing transactions with negative fees

matejcik, 2019-08-22:
one such transaction is in the second test vector: https://github.com/trezor/trezor-firmware/blob/master/tests/device_tests/test_msg_cardano_sign_transaction.py#L61-L85
where the input value is 3169260, output is 3003112 and change is 1000000, for a total of -833852
This then ends up wrongly formatted by `format_amount` which is not designed to work with negative values.

Presumably the network will reject such transactions, so the problem is not critical. Still, not great.

Care to take a look, @refi93 ?

prusnak, 2019-08-23:
Negative values in `format_amount` fixed in https://github.com/trezor/trezor-firmware/issues/439

mark-stopka, 2020-02-22:
So, is this fixed by #439 or it still needs additional work?

prusnak, 2020-02-22:
@mark-stopka no, #439 only fixes negative numbers formatting - the issue itself is still not fixed

gabrielKerekes, 2020-08-10:
@matejcik I think this can be closed now thanks to the Shelley update? Fee is no longer calculated from previous transactions, but is included in the transaction itself and it also can't be negative.

matejcik, 2020-08-10:
agreed

closed 2020-08-10 by matejcik