trezor/trezor-firmware #442: Refactor and polish SLIP-39 UI and storage

tsusanka, 2019-08-26:
Application layer:

- [ ] Improve storage.fetch function ([comment](https://github.com/trezor/trezor-firmware/pull/428/files/2eac710a0178a86c7a89fa1a59e91e4b83ca1faf#r316564892))
- [ ] Unify checklists ([comment](https://github.com/trezor/trezor-firmware/pull/428/files/2eac710a0178a86c7a89fa1a59e91e4b83ca1faf#r317530765))
- [ ] Checking words ([comment](https://github.com/trezor/trezor-firmware/pull/428/files/2eac710a0178a86c7a89fa1a59e91e4b83ca1faf#r317091811))
- [ ] Introduce a function for plural strings
- [ ] Move this check ([comment](https://github.com/trezor/trezor-firmware/pull/428/files/2eac710a0178a86c7a89fa1a59e91e4b83ca1faf#r317589964))
- [ ] Unify reset/recovery layouts (tracked in https://github.com/trezor/trezor-firmware/issues/409)
- [ ] refactor management.common.layout (connected to #409)

Tests:

- [ ] Move `enter_all_shares` to common
- [ ] Test all possible group+share comibinations in reset device test ([comment](https://github.com/trezor/trezor-firmware/pull/428/files/2eac710a0178a86c7a89fa1a59e91e4b83ca1faf#r317568026))
- [ ] Test all combinations in recovery test ([comment](https://github.com/trezor/trezor-firmware/pull/428/files/2eac710a0178a86c7a89fa1a59e91e4b83ca1faf#r317568601))

andrewkozlik, 2019-08-26:
- [ ] Skip threshold prompt if number of shares is 1.
- [ ] Discuss: Should we allow the group count to be 1 in advanced backup?

andrewkozlik, 2019-08-27:
- [ ] It would be nice to unify the naming of the constants and variables for advanced (group) shamir backup. Currently we have:
  - advanced_shamir true vs. false
  - Slip39_Single_Group vs. Slip39_Multiple_Groups
  - TYPE_SLIP39 vs. TYPE_SLIP39_GROUP

ciny, 2019-08-28:
@andrewkozlik 
1. created issue #463 
2. added as a topic for discussion to this epic. no issue yet
3. referenced in #459, if you have other suggestions regarding single vs multi group shamir naming please add it there

tsusanka, 2019-09-04:
Note to myself: Before doing a PR, I want to have a look at:
- [ ] _request_secret function
- [ ] mypy

closed 2019-09-26 by tsusanka