trezor/trezor-firmware #444: passphrase entry keyboard freezing

matejcik, 2019-08-27:
When entering passphrase, the keyboard will sometimes stop responding. After initiating a new action in the wallet, the device will start responding again.

The problem is reproducible, @bosomt can always do it.
The problem does not depend on passphrase length.

matejcik, 2019-08-27:
@bosomt, could you check if this is reproducible with the emulator?

bosomt, 2019-08-27:
@matejcik yes i was able to reproduce it via Emulator, come and see ;)

matejcik, 2019-10-01:
Lowering priorities and moving to backlog.
Let's check on this again when Wallet fixes the "device entered inconsistent state" problem, I suspect the root cause is the same.

sorooris, 2019-10-01:
https://github.com/satoshilabs/mytrezor/issues/848

matejcik, 2019-10-24:
the relevant Wallet issue has been fixed. @bosomt would you please retest this issue again?

bosomt, 2019-11-04:
i was not able to reproduce it for last 11 days.
as soon it happens again i will contact you

closed 2019-11-04 by bosomt