trezor/trezor-firmware #445: trezorctl command behavior for Super Shamir on T1

sorooris, 2019-08-27:
Nixos, Chromium
T1 BTC-only fw (1.8.3)
wallet Git Commit: 2fd93cffe03f290046355ff17a6dc55930324bef

**Steps to reproduce:**
- install latest fw
- open terminal and type in `trezorctl reset-device -e -t 256 -b shamir`
- observe

**Expected:**
Command for the device reset with Shamir groups is not allowed for T1.

**Actual:**
Using the command `trezorctl reset-device -e -t 256 -b shamir` on a wiped T1 generates a 24-word seed.

tsusanka, 2019-08-27:
I think we should solve this twofold:

1. Throw an error in trezorctl.
1. Add check to legacy.

What do you think @prusnak @matejcik?

matejcik, 2019-08-27:
it makes no sense to fix this in T1 firmware, as the bug will affect all non-updated Trezors anyway.

we will add the following warning to `trezorctl`:
> WARNING: Trezor One currently does not support Shamir backup.
> Traditional single-seed backup will be generated instead.

(let me know if you want to change the text)

But I believe that `trezorctl` should not try to *stop* you from sending the request with the appropriate field set.

matejcik, 2019-08-27:
I pushed a commit that adds the warning. Do we want to keep the issue open for further discussion?

tsusanka, 2019-08-27:
IMHO better closed

matejcik, 2019-09-13:
the code should be changed to use capabilities instead of guessing based on model number

matejcik, 2019-09-17:
right now this depends on #518

commit 3e90c7fe (2019-08-27): python/trezorctl: add warning when initializing a T1 with Shamir

fixes #445
commit 8b4db0f0 (2019-09-19): trezorctl: use capabilities to detect Shamir backup (fixes #445)
closed 2019-09-19 by matejcik