trezor/trezor-firmware #449: U2F/WebAuthn Device Tests

prusnak, 2019-08-27:
Create Device Tests for U2F/WebAuthn

prusnak, 2019-09-14:
U2F tests merged to CI via https://github.com/trezor/trezor-firmware/commit/650db2a5c01b80cc745d436f5f4d52b9cc52a93a

andrewkozlik, 2019-10-08:
Pull requests: https://github.com/solokeys/fido2-tests/pull/19, https://github.com/trezor/trezor-firmware/pull/606

andrewkozlik, 2019-10-17:
The pull request has been merged into solokeys/fido2-tests. We now need to integrate this into the CI. @tsusanka the steps to get the tests running are as follows: 

1. The tests require the `fido2` package. I wasn't sure whether to add that to the Pipfile in trezor-firmware or whether that should go into some separate Pipfile file in the CI.

2. Clone git@github.com:solokeys/fido2-tests.git.
Note: If solokeys make some updates to the tests which are incompatible with Trezor, they might break our CI. So we should check out a fixed version or maybe create a fork of this repo and use that in the CI, so that we have control over when the changes get introduced, or add this as a submodule to trezor-firmware. I am not sure what the standard solution here is.

3. Launch the tests by running `pytest --sim tests/standard/ --vendor trezor` in the `fido2-tests` root directory.

prusnak, 2019-10-17:
I completed 1) and 2) in bd79827c524b65951c46ca22b2c255abe653fec7

I leave 3) to @tsusanka

tsusanka, 2019-10-18:
I have added the tests to CI in b530e7c1053f6ddcd9929ec1f71ba25a0d586eaf. However, it seems to me that the test suite is failing when the randomization is turned on. @andrewkozlik didn't you have a similar experience? I've solved it for now by setting it to [some seed](https://github.com/trezor/trezor-firmware/commit/b530e7c1053f6ddcd9929ec1f71ba25a0d586eaf#diff-7fcd8f4a5caeaad75fdeaca6b937a07dR35), which passes, but is it something we care to fix?

prusnak, 2019-10-18:
Decreasing priority, because we have the test in master now. We just need to find out why random ordering is bad.

prusnak, 2019-11-09:
Closing. Let's investigate the random ordering via https://github.com/trezor/trezor-firmware/issues/682

closed 2019-11-09 by prusnak