trezor/trezor-firmware #450: implement CoSi on Trezor T

matejcik, 2019-08-27:
The CoSi signing method is available on T1 (so it exists in trezor-crypto), but not on TT. It should be added at some point.

matejcik, 2020-01-07:
related, what occured to me when working on keyctl:

perhaps the sequence of CosiCommit and CosiSign should be considered a single operation with only one confirmation screen? Namely, if a CosiCommit is followed by a CosiSign with the same `address_n` and `data`, it could be considered a part of the same flow.

This however depends on exact security properties of the CoSi scheme.

1) Is confirmation for the commitment even required? Can an attacker abuse the commitment somehow?
2) Does it make sense to confirm commitment and signature separately? Does it make sense to want to commit to a digest and later decide not to sign it?
3) If (1) is yes, and (2) is no, could we use the commitment as a token that allows prompt-less signing?
4) Can we somehow check the validity of `global_commitment` and `global_pubkey`? Can an attacker substitute malicious values to obtain something interesting?

matejcik, 2022-11-10:
in light of the CoSi vulnerability, the flow should look as follows:

1. there is only *one* entry point, i.e., `CosiCommit`. `CosiSign` cannot be started separately.
2. generate a nonce and store it in sessionless cache. if there is already a cached nonce, reuse it.
   * reason: otherwise an attacker could pump out entropy via repeatedly CosiCommiting. this is _prooobably_ not a vulnerability but we disallow it just in case
   * sessionless cache to save memory; the CoSi flow can't work across sessions anyhow
4. send the commitment and wait for a `CosiSign`
5. load the cached nonce into a local variable and erase it from cache
6. show confirmation screen
7. produce and send the signature

matejcik, 2022-11-10:
3.5: check that the `address_n` in `CosiSign` is the same as in `CosiCommit`, abort otherwise

prusnak, 2022-11-10:
BTW, if we implement CoSi on Trezor T, we can drop the code completely from T1.

closed 2022-12-09 by mmilata

prusnak, 2022-12-11:
> BTW, if we implement CoSi on Trezor T, we can drop the code completely from T1.

Followup PR in https://github.com/trezor/trezor-firmware/pull/2675 but we are not merging this until this PR has been successfully tried in the wild (as in we were indeed able to sign the FW release with Model T).

PR #2662 (2022-12-09): core: CoSi collective signatures
Fixes #450.
[[UI diff](https://satoshilabs.gitlab.io/-/trezor/trezor-firmware/-/jobs/3382048498/artifacts/master_diff/index.html)]

[image]