trezor/trezor-firmware #451: SuperShamir: Remaining shares are very confusing and probably incorrect

tsusanka, 2019-08-28:
1. Why is the screen displayed _before_ the Success screen? It is in this order:

[image] [image] [image]

I believe we should first inform the user, what they've entered and then what is remained.

2. I'm not sure this works correctly. I've entered G2, S2 and then G1, S1 in 2of3 groups scenario, where each group is 2of3. After I enter the second share I get:

[image] [image]
[image]

It seems to me the Remaining shares do not take into account the last entered share.

---

Also, I'm not sure it is clear to the user that the screen displays what is left, but not all of that is needed to be entered.

andrewkozlik, 2019-08-28:
Some thoughts/nitpicks with regard to these screens:
- I am not sure I like "Share 1 from Group 1" being split across three lines if it can fit on one line. If we really want it to be split, then perhaps center-align the three lines.
- How about putting an ellipsis (...) on the line following the last word? Like this: 
  2 more groups starting
  gesture
  negative
  ...
  Any opinions?
- How about making the Success title and/or checkmark green? First please check some other success screens, so that the formatting is consistent.

tsusanka, 2019-08-28:
> How about making the Success title and/or checkmark green? First please check some other success screens, so that the formatting is consistent.

The Success page should have green checkmark. This indicates it does not use the common.show_success page as it should.

ciny, 2019-08-28:
I just went through the whole process (on the release/2019-09 branch)
- the screens were in the correct order (enter words->share success->homescreen->remaining shares->enter words->share success etc) including restarting the emulator between shares to test persistence. I was not able to reproduce this behavior
- I was shown the correct remaining shares. The only thing that might be confusing is that we offer "x more groups" even in case we already started enough groups to reach the group threshold. In this case we probably shouldn't display the "x more groups" page at all and just show the already started groups that haven't reached their threshold.

@sorooris please retest on the release firmware so we can be sure it's not some blocker that needs addressing in the release branch. Shares that already reached their threshold being shown is imho a dealbreaker

@andrewkozlik I made the screens according to the designs provided by our designers. @ZdenekSL is probably the one to approve or communicate changes in the UI. 

anyway seems the possible blocker parts of this issue can't be reproduced so we can discuss the rest of the suggestions at standup and solve them in #442

tsusanka, 2019-08-28:
The issue is a bit different. We correctly show success pages after entering the words, but clicking on `Enter share` leads to the info about Remaining shares, which is really weird and confusing.

tsusanka, 2019-09-04:
What we need to discuss with UX is this UX screen:

[image]

which couldn't be implemented because we need also the Wipe button. This caused this whole confusion. To be discussed IRL

tsusanka, 2019-09-10:
Closed in favor of #512.

closed 2019-09-10 by tsusanka