trezor/trezor-firmware #453: Dry-run - cancelling the process after selecting wrong word number

sorooris, 2019-08-28:
Nixos, Chromium
TT 2.1.5 (fd53c72)
wallet Git Commit: 2fd93cffe03f290046355ff17a6dc55930324bef

_Precondition: the device is using for example a 12-word seed_
**Steps to reproduce:**
- initiate dry-run in the Trezor wallet
- select a word number other than the correct one on the device screen
- enter any words and finish the dry-run
- observe
 
**Expected:**
The device screen allows the user to either try again or cancel the dry-run process.
 
**Actual:**
The device screen displays only a "Try again" option without a comfortable option to cancel the process and is forced to disconnect the device and after reconnecting the dry-run persists but allowing to cancel the process from there.
Wallet then throws error messages (Unexpected message) both on screen and in the console if the _remember device_ option is selected.

**Screenshots:**
[image]
[image]
[image]

tsusanka, 2019-08-28:
If this is the first share entered, we should prompt the user to enter the number of words again. If this is too complicated, I'm inclined to not doing it.

sorooris, 2019-08-28:
@tsusanka It is indeed the first share entered, incorrectly (any random words since there is no option to go back or cancel), after selecting a Shamir backup with 20 words instead of a single backup with 12 words as described above.

tsusanka, 2019-09-03:
PR available. Fixed for normal recovery as well, not just dry run. I hope the UX is not confusing.

closed 2019-09-05 by tsusanka