trezor/trezor-firmware #459: Unify the usage of single and multiple group shamir

ciny, 2019-08-28:
Unify the usage of single vs group shamir - I believe we should have a single API for that, which works with the groups. This is up for discussion and suggestions on how to best approach it.

ciny, 2019-08-28:
[suggestion by andrew](https://github.com/trezor/trezor-firmware/issues/442#issuecomment-525265910)

ciny, 2019-08-28:
From discussions we already have:
- slip39 is probably preferred over shamir as prefix when it comes to code itself. It will be consistent with bip39 and external developers will be imho less confused when they put slip39 instead of shamir into a search engine
- suffix is still up for discussion. IMHO we should follow (and ideally use) ResetDeviceBackupType enum wherever we can. the enum values (0, 1, 2 for bip39, single slip and group slip respectively) are consistent between mnemonic.py and ResetDeviceBackupType so it's just a matter of naming. The only issue I see with this is that the names in BackupType are pretty long and if we apply this pattern to function naming it can make for some pretty ugly function names. But hopefully if we refactor the layouts there should be only few points where we will make a difference between various types.

tsusanka, 2019-09-09:
Let's use `BackupType.{Bip39, Slip39_Basic, Slip39_Advanced}` everywhere and the naming basic vs advanced everywhere as well.

tsusanka, 2019-09-18:
@ciny still something to be done here?

ciny, 2019-09-18:
Just went through the code. The general idea of this is done. We probably should rename trezor.ui.shamir to trezor.ui.slip39 so we use slip39 terminology everywhere in the python part of codebase.

tsusanka, 2019-09-18:
Yes, let's rename everything we can from shamir to slip39. We should also consider renaming Capabilites from Shamir to Slip39_{Basic,Advanced}, but I'm not sure @prusnak will be fond of that.

@ciny may I leave the this one to you?

closed 2019-09-18 by tsusanka