trezor/trezor-firmware #460: Check Min/max allowed values for various reset device settings

ciny, 2019-08-28:
Check the min/max allowed values (number of shares, groups, and threshold for each) for various reset scenarios (bip39, slip39 single, slip39 group). Related to #455

tsusanka, 2019-09-18:
@ciny is this still a thing?

ciny, 2019-09-18:
max value for number of groups/shares is 16
max value of group/share threshold is the number of groups/shares
min value for number of shares is 1 in case of advanced slip39 and 2 in case of basic slip39
min value for share threshold is 2 unless the number of shares is set to 1 in which case it's 1
min value for number of groups is 2
min value for group threshold is 1 

these are the current constraints. AFAIK that should be correct.

closed 2019-09-18 by tsusanka