trezor/trezor-firmware #463: Skip Threshold prompt if number of shares is 1

ciny, 2019-08-28:
If number of shares is 1 then the threshold can also only be one, in that case we can skip this step. We have to make sure it won't be confusing for the user so UX will be needed

ZdenekSL, 2019-09-05:
We decided that we will hide the screen with threshold if the shares are set to 1 in the group.

tsusanka, 2019-09-11:
I've implemented this, but I think it's really not a good idea. Example workflow, I start Super Shamir and:

- I set number of shares for group 1 and choose 5 shares.
[image]
- I set threshold for group 1 and choose 3.
[image]
- I set number of shares for group 2 and choose 1 share.
[image]
- Now I am supposed to choose number of shares for group 3. I did not get any confirmation or anything that the threshold is equal to 1 and it is really easy to not realize that I'm setting the number of shares for group 3 instead of the threshold.
[image]

I think this does more harm than good.

tsusanka, 2019-09-11:
Also note, that we already have a screen where you can't set anything. And that's when number of shares is 2, we enforce the threshold to 2 (not in groups threshold):

[image]

tsusanka, 2019-09-18:
After discussion with @matejzak we have decided not to implement this, because we believe it confuses people. Petr (UX) seem inclined to leave it as well.

closed 2019-09-18 by tsusanka

sorooris, 2019-09-18:
@tsusanka what is the current option then?

tsusanka, 2019-09-19:
We will have a screen with the threshold even it is equal to 1 and it can't be changed.

[image] [image]