trezor/trezor-firmware #468: Unifty core emulator running scripts

prusnak, 2019-08-30:
Currently, we have:

* `emu.sh`
* `trezor_cmd.sh`
* `tests/run_tests.sh`
* `tests/run_tests_device_emu.sh`
* `tests/run_tests_device_emu_monero.sh`

and this is becoming messier over time.

We should refactor/unify these scripts into something sane.

jpochyla, 2019-09-03:
It would be convenient to wrap/rewrite monero tests into pytest.

prusnak, 2019-09-16:
+1 for rewriting these into Python.

Also pytest can accept an option `--emu path/to/emulator` which will start an emulator if needed (before the whole test suite, or before each test separately)

ZdenekSL, 2019-10-22:
Related to #466

malinah, 2019-11-04:
File `trezor_cmd.sh` is actually not a running script but rather a bash library for the TT emulator. That was my attempt to unify the control over the TT emulator to fit the existing testing environment and contains handling of features that are not covered elsewhere. So, please take it into account when the whatever is going to be rewritten. Thanks.

tsusanka, 2019-12-09:
Let's do this after #466

commit f838b78a (2020-01-16): core: use better emulator test wrappers (fixes #468)
commit 4e94fcb3 (2020-01-16): core: use better emulator test wrappers (fixes #468)
commit ce0353bf (2020-01-17): core: use better emulator test wrappers (fixes #468)
commit ec192e94 (2020-01-24): core: use better emulator test wrappers (fixes #468)
closed 2020-01-24 by matejcik