trezor/trezor-firmware #471: Add device test for PIN change on Trezor T

andrewkozlik, 2019-08-30:
- Make it possible to enter a PIN into the Trezor T in device tests.
- Add device test for change PIN on Trezor T.

prusnak, 2019-09-02:
Currently, we cache the entered PIN on T1 in order to check whether the PIN was set correctly.

The question is whether we need to do the same for TT or there is a clever trick to perform this check differently (e.g. by returning PVC + all metadata required for checking whether the PIN corresponds to the currently tested PIN). If there is such trick I suggest to remove PIN caching in T1 and use the same method for both T1 and T2.

matejcik, 2019-09-09:
we could call change-pin again, and check that the new pin is accepted as "current pin"

tsusanka, 2019-09-20:
Looks like a duplicate of #228?

prusnak, 2019-09-20:
Closed via https://github.com/trezor/trezor-firmware/commit/65c1afa728ada4700f4afb710187300560e14079

closed 2019-09-20 by prusnak