trezor/trezor-firmware #482: Binance: Fix BNB Divisibility

notatestuser, 2019-09-03:
Hi,
The following setting appears to be incorrect:

https://github.com/trezor/trezor-firmware/blob/33cc147c66c98885afaf95f7a267db19c37a3823/core/src/apps/binance/helpers.py#L21

All tokens on Binance Chain (including BNB) have a fixed decimal scale of 8.

Reference: https://docs.binance.org/faq.html#what-is-the-decimal-precision-for-prices-and-quantities-on-binance-chaindex

tsusanka, 2019-09-03:
@ciny in our code you are linking https://www.binance.vision/glossary/jager but that reference seems to concur with the link above as far as I can tell.

ciny, 2019-09-03:
@tsusanka I'm pretty sure the value was 18 at the time, maybe the page was referring to the ERC20 token, idk. Either way this is a simple fix.

closed 2019-09-03 by tsusanka