trezor/trezor-firmware #487: Redesign DebugLink

prusnak, 2019-09-03:
We might want to redesign how DebugLink works. It was designed during the T1 development, but lots has changed since then. Let's revisit how it works and maybe do some changes:

Ideas:
* introduce new messages that allow to perform click on TT touch display: (x,y) coordinates
* make protocol usable with tools such as [tpmb](https://github.com/mmahut/tpmb/)
* remove unnecessary messages

tsusanka, 2019-09-24:
Bumping the priority again because of #562, which does not have a test because of this.

ciny, 2019-09-24:
#547 and #557 are also blocked by this.

matejcik, 2019-09-24:
introductory thoughts:

The general problem with input automation is detecting that the input was processed. This is currently solved in two ways in TT:

1. Same as in T1, most discrete steps are acknowledged by a ButtonRequest from the device. This allows the host to see how far along the device is in general, which is useful even for Wallet interface, etc. As the input complexity grows, however, this gradually becomes insufficient, see also #35, #232 

2. With "input event" level granularity, TT has an implicit input event queue. This means that in theory, we could blast the whole sequence of inputs at once, and TT would just consume them at the appropriate places. Unfortunately, that means that there is no feedback, so we can't detect failure. (see test in #560)

Plus, in places where debuglink returns a response (e.g. reading mnemonic words), the appropriate code can wait until the response becomes available.

---

We could write tests that simply do a `sleep(1)` after every input event. It's quite stupid, and it still doesn't give us a good way to detect failure, but we could do it as a stopgap measure.

Every layout change could trigger a debuglink response; say, the host sends a DebugLinkAwaitLayout, every UI element is instrumented to respond to this, and the debuglink call returns after the UI has changed.

This could be extended further and the layout could respond with rich information, for starters, the type (Text, Info...) and text content. This would let us directly write assertions on screen contents, which would be super nice.

closed 2020-03-27 by tsusanka