trezor/trezor-firmware #491: Binance: Improved decimal formatting

notatestuser, 2019-09-03:
This is my final UX-related suggestion :)

On Binance Chain, all tokens have a [decimal scale of 8](https://docs.binance.org/faq.html#what-is-the-decimal-precision-for-prices-and-quantities-on-binance-chaindex) and amounts are represented as integers in transactions. This means that an integer value of `100000000` in a transaction represents an amount of `1.00000000` in decimal form.

It would be nice to display amounts in this decimal format when they are displayed on the device screen. Ideally, this would apply to both Transfers and Orders.

tsusanka, 2019-09-03:
@ciny what do we display now?

prusnak, 2019-09-03:
@notatestuser please prepare and send a pull request

prusnak, 2019-09-04:
Implemented in https://github.com/trezor/trezor-firmware/pull/494

PR #494 (2019-09-13): Binance UX improvements
closes #490 
closes #491
closed 2019-09-13 by ciny