trezor/trezor-firmware #500: Enable 1 of 1 Shamir seed

ZdenekSL, 2019-09-05:
Let's enable creating 1 of 1 when creating new a wallet with Shamir backup

tsusanka, 2019-09-09:
To clarify: we have agreed we will allow 1of1 share in a Basic Slip39 (meaning no groups). Is that correct?

tsusanka, 2019-09-11:
We probably should retext this screen then. Maybe we could inform the user that it is very similar to Single Backup?

[image]

tsusanka, 2019-09-11:
@matej_zak could you have a look?

tsusanka, 2019-09-12:
@matejzak do you think you could come up with some better text?

tsusanka, 2019-09-18:
Suggestion: `Only one share will be created.`

tsusanka, 2019-09-18:
Note: we should also fix the possible duplicates in the "words checking" because it is way more likely to happen now (academic, academic).

tsusanka, 2019-09-20:
Duplicates fixed in  https://github.com/trezor/trezor-firmware/pull/541

PR #546 (2019-09-20): core: enable 1of1 in reset device
Enables 1of1 in reset device and alters the texts accordingly.

Closes #500
closed 2019-09-20 by tsusanka