trezor/trezor-firmware #512: Add Info button to Recovery homescreen for Super Shamir

tsusanka, 2019-09-10:
In Recovery Homescreen in case we already know it is a Super Shamir, let's display an additional "Info" button, which will show, which groups and shares are needed.

Unfortunately we are using the Confirm dialog, which works with two buttons only, so this will need some refactor.

Fast prototype:

[image]

tsusanka, 2019-09-10:
Clicking on the Info button will lead to:

[image]

andrewkozlik, 2019-09-11:
Note about the screenshot: "More shares needed for this recovery" should instead say something like "More shares needed for seed check"

tsusanka, 2019-09-11:
Good catch. I've removed the second part completely, I think it's fine.

[image]

tsusanka, 2019-09-16:
Although inconsistent I think "Abort" is clearer than the cross:

[image]

ciny, 2019-09-17:
To test this screen we will need to modify the debuglink so we can press the info button.

matejcik, 2019-09-17:
> To test this screen we will need to modify the debuglink so we can press the info button.

let's defer that to #487

tsusanka, 2019-09-17:
> let's defer that to #487

Agreed.

ciny, 2019-09-17:
implemented in #524

PR #524 (2019-09-17): core: add InfoConfirm
This adds another Confirm dialog with an Info button and closes #512.

Although the Confirm dialogs bloat a little I think we might use the Info button somwhere else as well. Or we refactor this as soon as we have a new styleguide.
closed 2019-09-17 by ciny