trezor/trezor-firmware #521: Binance U2F registration forgotten after FW upgrade to 2.1.5

brianddk, 2019-09-13:
## Symptoms

When a Trezor-T is registered as a U2F device on `binance.com` the registration is forgotten (by binance) when the Trezor-T is upgraded from 2.1.1 to 2.1.5.  Interestingly enough, the Trezor-1 at FW level 1.8.3 will be "recognized" (by binance) in place of the Trezor-T.

## Reproduction

1. `python3 trezorctl firmware-update -v 2.1.1`
2. Register Trezor-T as your U2F device on `binance.com`
3. Confirm U2F based 2FA functionality on `binance.com`
4. Logout and clear cache related to `bianance.com`
4. `python3 trezorctl firmware-update -v 2.1.5`
5. Attempt U2F based 2FA login on `binance.com`
5. Observe that `binance.com` spins waiting for registered U2F device
6. Abort login, clear cache and insert Trezor-1 device
4. `python3 trezorctl firmware-update -v 1.8.3`
5. Attempt U2F based 2FA login on `binance.com` with Trezor-1
6. Observe success with Trezor-1 where Trezor-T failed

## Expected behaviour

Once a Trezor (1 or T) has been registered as a U2F devices, that registration should remain valid independent of FW flash level.

prusnak, 2019-09-13:
Honestly, this report is not very useful, because of the interference with the T1 device. Can you try to reproduce again without adding T1 device into the mix?

andrewkozlik, 2019-09-13:
I am unable to reproduce this using Chromium 76.0.3809.100 on 64-bit Ubuntu 18.04. According to the steps given above, the bug occurs before the T1 comes into the mix, but is that really what happened? The problem with using two devices for U2F with the same seed is that the U2F counters will get mingled. I tried to manually decrease the U2F counter and found that upon attempting to authenticate with the Trezor T, Binance detects this and shows "2FA verification code error". When I increase the counter back to the original value, it starts working again. (To be clear, this behavior has nothing to do with the firmware update and there is nothing we can do about it, since it's in accordance with the specs.)

brianddk, 2019-09-13:
> Honestly, this report is not very useful, because of the interference with the T1 device. Can you try to reproduce again without adding T1 device into the mix?

Unfortunately not.  `binance.com` is now geo-blocked from my region.  I'll see if I can evade the geo-blocking over the next few days.

brianddk, 2019-09-13:
> U2F counters will get mingled

@andrewkozlik, its my understanding that the U2F counter is always set to "unix time epoch" which is specifically designed to clear up the counter mingling you reference.   I'll look at jacking with the counters to see if that clears it on my end.

andrewkozlik, 2019-09-14:
> @andrewkozlik, its my understanding that the U2F counter is always set to "unix time epoch" which is specifically designed to clear up the counter mingling you reference. I'll look at jacking with the counters to see if that clears it on my end.

The U2F counter gets set to the current UNIX time only during device recovery from seed. During normal use it just gets incremented by 1 every time.

brianddk, 2019-09-14:
@andrewkozlik, @prusnak, `python3 trezorctl set-u2f-counter 1568484984`  fixed the problem.  Thx for the suggestion.

Closing issue.

closed 2019-09-14 by brianddk

GreatBigWhiteWorld, 2019-11-10:
What does "python3 trezorctl set-u2f-counter 1568484984" mean? How to get the magic number "1568484984" to solve the problem?

prusnak, 2019-11-10:
@GreatBigWhiteWorld it's just the current timestamp - available from here for example: https://www.unixtimestamp.com/

GreatBigWhiteWorld, 2019-11-10:
Thank you. Though I still don't understand how it solved his problem. Did the firmware update break the counter's time counting? So he reset it to the then current time. He also need to be very accurately issue the time-reset-command at that exact second when the command was issued?

prusnak, 2019-11-10:
By setting the counter to current time you set it to a value that is guaranteed to be higher than the whatever other value was there before. Which fixes the problem.

GreatBigWhiteWorld, 2019-11-10:
OK. So this counter is not necessarily to be accurate but needs to have a higher value than the previous one. I guess there is another counter that use real unixtimestamp since the service side uses real unixtimestamp along with the shared secret to generate the code?

prusnak, 2019-11-10:
> So this counter is not necessarily to be accurate but needs to have a higher value than the previous one.

yes

> I guess there is another counter that use real unixtimestamp since the service side uses real unixtimestamp along with the shared secret to generate the code?

no

GreatBigWhiteWorld, 2019-11-13:
I know it is not the place to ask, but can you give me some material to read about this TOTP 2fa mechanism that I'm missing? To my understanding, you need unixtimestamp with an additional same secret for both the client and the service side to generate the same code? If trezor picks a random larger number, it can generate a code that the service side still confirms?

andrewkozlik, 2019-11-13:
The TOTP spec is [RFC 6238](https://tools.ietf.org/html/rfc6238). We have a ticket for supporting TOTP in Trezor (https://github.com/trezor/trezor-firmware/issues/33) but the problem is that Trezor does not have an internal clock and there is no standard way of getting the current time via USB, unless you have a special application running on the host (this could be added to Trezor Bridge).

GreatBigWhiteWorld, 2019-11-15:
OK. So it's currently not TOTP in the first place. I might get it. Thanks.