trezor/trezor-firmware #535: Add upgrade test for U2F counter

prusnak, 2019-09-18:
We should check the U2F counter before and after the upgrade stays the same.
We can't read the U2F counter from the storage directly, but we can use U2F/FIDO to perform two authentications and check the counter was exactly one higher than the last one.

tsusanka, 2019-11-11:
I wanted to do this naively by introducing `u2f_counter` in DebugLinkState. However, we still need to do some u2f action to set the counter otherwise it is None and to check if None persisted is not exactly what we want.

matejcik, 2019-11-11:
shouldn't the U2F counter be set at reset/recovery time?

prusnak, 2019-11-11:
My idea was to just call SetU2FCounter message (or set it via RecoveryDevice like matejcik suggests).

tsusanka, 2019-11-12:
Yes, calling SetU2FCounter is a good choice. What I'm wondering is how to receive the value. We can either:

1. Return the value in DebugLinkState. Get the value there by introducing a getter for the U2F counter in storage.
2. Return the value in DebugLinkState. Get the value there by calling `next_counter` each time we call for DebugLinkState.
3. Introduce a new message DebugLinkGetNextU2FCounter and return the value in a new message DebugLinkNextU2FCounter.

(1) seems ugly, since we would add some production code just because of this test. (2) seems a bit stupid, because we would increase the counter every time we call for debug state. On the other this is a _debug_ app, so maybe we do not care? I'm not sure about the ramifications of this for U2F tests for example. Although (3) is the biggest option code-wise, it seems the most correct one to me.

matejcik, 2019-11-12:
> (1) seems ugly, since we would add some production code just because of this test.

what "production code" do you mean?

tsusanka, 2019-11-12:
> what "production code" do you mean?

I mean we would be adding a getter for U2F counter, which has no "regular" usage throughout the code. It is only because of tests.

matejcik, 2019-11-12:
Oh, internal code. Ok. It might not even be possible to get the current value though? The counter is configured for read-and-increment behavior.

Anyway I have option (4): extend `DebugLinkGetState` with `bool read_u2f_counter`, and call `next_counter` only when this field is set. A little less work than (3).
Question is, is it better to endlessly expand `DebugLinkState`, or make a new message?

Also maybe `GetNextU2fCounter` could be public too?

tsusanka, 2019-11-12:
> It might not even be possible to get the current value though? The counter is configured for read-and-increment behavior.

Exactly. It's more work than I have expected.

> Anyway I have option (4): extend DebugLinkGetState with bool read_u2f_counter, and call next_counter only when this field is set. A little less work than (3).

I like that! In general I'm for new messages, but it would be better to split the current State as well. So maybe it's fine for now?

> Also maybe GetNextU2fCounter could be public too?

I honestly don't know.

---

I think (4) is the best option for now.

tsusanka, 2019-11-12:
We have agreed on standup GetNextU2fCounter is a way to go.

closed 2019-11-13 by tsusanka