trezor/trezor-firmware #542: unit tests for process_slip39

matejcik, 2019-09-19:
The function `recover.process_slip39` is untested, and relies on some check being done by the UI.

In particular, it is unclear what would happen if the UI checks failed and the user entered a Slip39 Basic mnemonic first, and a mnemonic *with the same first 2 words* but from a different set second.

There should be unit tests to verify its behavior.

tsusanka, 2019-09-23:
Do I understand it correctly that because unit tests are run using micropython itself, we can't use `mock` package because it seems there is no port for micropython, is that correct? I didn't find it in [micropython-lib](https://github.com/micropython/micropython-lib) nor did I get any reasonable google search results.

The reason I'm asking is that I need to mock the storage interface. I guess I can do it the simple way where I redefine the functions with some simple lambdas, but it is significantly less powerful than `mock`.

cc @prusnak

matejcik, 2019-09-23:
> Do I understand it correctly that because unit tests are run using micropython itself, we can't use `mock` package because it seems there is no port for micropython, is that correct?

yes.
we can implement a simple Mock object in ~30 lines that should be good enough for our purposes. an implementation is likely to lie around on StackOverflow somewhere :)

tsusanka, 2019-09-23:
I've made a separate issue for that: #559. Let's do this issue after that one then.

tsusanka, 2019-10-03:
> In particular, it is unclear what would happen if the UI checks failed and the user entered a Slip39 Basic mnemonic first, and a mnemonic with the same first 2 words but from a different set second.

@matejcik so you mean first the user enters `extra extend academic ..` which is Slip39 Basic and then `extra extend romp`, which has the same identifier, but different group settings.

tsusanka, 2019-10-03:
It indeed seems to behave unexpectedly. I'll look into it.

matejcik, 2019-10-03:
draft PR: #592

PR #601 (2019-10-04): Add unit test for process_slip39 function
Adds unit test for `process_slip39` function. Note that it still can happen that an unfitting mnemonic is stored in the storage and then later combined together with valid ones in `combine_mnemonics`. In such case, the MnemonicError is thrown and later caught in `_continue_recovery_process` and an error is shown. I believe that's an expected behaviour since this can happen very rarely, I doubt it can happen under "real-life" circumstances.

An example of this when the user enters a second mnemonic, which has the same identifier, but different member th [... cut]
closed 2019-10-04 by tsusanka