trezor/trezor-firmware #550: firmware 2.1.6 (702f865e) Firmware error when trying to create wallet with Single Backup (12 words)

pankrats, 2019-09-20:
Trezor T with 2.1.6 (702f865e). 

Steps to reproduce: 
1. Connect a wiped device, open wallet.trezor.io or beta-wallet.trezor.io
2. Select Create new wallet -> Single Backup, confirm on device
3. Click on Create backup in 3 minutes, Continue

Result: first Firmware error, then Backup failed in the wallet

[image]
[image]

matejcik, 2019-09-20:
the FW seems to think that backup is already done.

matejcik, 2019-09-20:
no, i was wrong
```
Traceback (most recent call last):
  File "/home/matejcik/projekty/prace/trezor-firmware/core/src/trezor/wire/__init__.py", line 319, in handle_session
  File "/home/matejcik/projekty/prace/trezor-firmware/core/src/apps/management/backup_device.py", line 19, in backup_device
  File "/home/matejcik/projekty/prace/trezor-firmware/core/src/apps/management/reset_device/__init__.py", line 203, in backup_seed
  File "/home/matejcik/projekty/prace/trezor-firmware/core/src/apps/management/reset_device/__init__.py", line 157, in backup_bip39
ValueError: Invalid data length (only 16, 20, 24, 28 and 32 bytes are allowed)
```

matejcik, 2019-09-20:
importantly, we do not have a test for this!

`reset_device(skip_backup=True)` then `backup_device()`

prusnak, 2019-09-20:
Please don't close before adding a test case for this

prusnak, 2019-09-20:
This can be also reproduced without using wallet (just press skip on the touch screen during normal reset).

matejcik, 2019-09-20:
the reason is that the common method `backup_seed` takes the secret in one case (when run from reset-device) but the text of the mnemonic in the other (when run from backup-device)
i'm now trying to find out where this happened.

matejcik, 2019-09-20:
seems to be introduced in e6372ede75fe03792a701b4e6d8e3d8cac31eca0

prusnak, 2019-09-20:
Delayed backup seems to work with Shamir and Super Shamir but can we have also test for these as well?

matejcik, 2019-09-20:
I have a fix ready, still making the test

prusnak, 2019-09-20:
Tests will be tracked in https://github.com/trezor/trezor-firmware/issues/551

Fix submitted to `master`:
* https://github.com/trezor/trezor-firmware/commit/78ecc38b1b1f5f3e547bcd95f3b33b1a1012a402
and cherrypicked to `release/2019-10`:
* https://github.com/trezor/trezor-firmware/commit/f6168b82465e12b5849cb1addfff03e31db1d978

closed 2019-09-20 by prusnak