trezor/trezor-firmware #552: Idea: TLS termination in the device

prusnak, 2019-09-20:
We could have TLS termination in the device (to show info served from HTTPS endpoints) if we had root CAs bundled with the firmware.

Such list is available from Mozilla for example: https://hg.mozilla.org/mozilla-central/raw-file/tip/security/nss/lib/ckfw/builtins/certdata.txt

matejcik, 2019-09-23:
the full list of roots is rather big though; on my PC, 800 kB in base64. We would need to pick-and-choose, which seems like a bad idea.

prusnak, 2019-09-23:
We are not going to support old RSA certificates, only ECC. This makes the selection much smaller.

Most of the services in crypto space use Cloudflare and they only support ECC certs in their free offering.

tsusanka, 2019-09-23:
> We would need to pick-and-choose, which seems like a bad idea.

In case we hard code the website(s), which I'm not sure we want, how bad idea is it actually? We could add only such CAs we need for that website. Of course this would break in case the website/CDN changes its CA provider, but I'm wondering if that isn't an acceptable risk for non critical features.

tsusanka, 2019-09-23:
> Thirdly, we can have the certificates on the SD card where they would be in a signed file, so Trezor could check the signature.

Better idea.

tsusanka, 2020-09-17:
We want to define this in Product first.

closed 2020-09-17 by tsusanka