trezor/trezor-firmware #558: Shamir recovery homescreen message handling

matejcik, 2019-09-23:
The recovery homescreen is persistent, in that when you disconnect and reconnect Trezor, you get back to the recovery homescreen.

After disconnect/reconnect, Trezor does not accept most message types. This is problem (A): when you plug in a trezor-in-recovery, you can't talk to it via USB. Related: #530, #232, #533

*Before* disconnect/reconnect, however, Trezor does accept other message types. This is problem (B): other messages will trigger an exit from the recovery homescreen, but do not clear the persisting status.

In particular, on a wiped Trezor, repeated call to RecoveryDevice will kill the homescreen and will not be able to return to it.

tsusanka, 2019-09-23:
Maybe we should do a "restart" after the recovery is started?

matejcik, 2019-09-23:
idea to resolve both: make recovery_device homescreen "reentrant".

RecoveryDevice will always trigger the recovery homescreen. Instead of [this error](https://github.com/trezor/trezor-firmware/blob/master/core/src/apps/management/recovery_device/__init__.py#L65), we make sure that the call goes through. The worst that can happen is that the user is thrown back to the homescreen when they're in the middle of entering a word; this is acceptable, IMO.

We register RecoveryDevice in the persistent mode too, hence fixing problem (A) -- and we let the user go back to recovery in situation (B).

There is still part of (B) open. The restart -- or maybe just re-registration of USB messages? -- will probably resolve that.

tsusanka, 2019-09-23:
> There is still part of (B) open. The restart -- or maybe just re-registration of USB messages? -- will probably resolve that.

If the loop is empty, the device reboots itself. So we could do the the same thing as we do to get out of the persistence. See [here](https://github.com/trezor/trezor-firmware/blob/master/core/src/apps/management/recovery_device/homescreen.py#L25).

In other words, in `recovery_device` after receiving RecoveryDevice msg we could set the `set_in_progress` to True and clear the loop, rebooting the device respectively. Therefore we would not call the `homescreen.recovery_process` from the main function, but we would invoke the persistence mode using the reboot.

This would break, however, the possibility to do the whole recovery with wire.Context and receiving a Success message at the end. On the other hand maybe that is desirable, because we would remove this inconsistency and it does make sense.

matejcik, 2019-09-23:
the result of a RecoveryDevice should be an immediate Success: "device has entered recovery mode"

tsusanka, 2019-09-24:
> the distinct problem with this is that it makes it impossible for the wallet to display results of dry-run

Although that is true, I'm not sure this is a problem. If you replug the device during the dry-run the wallet won't display the result as well. Yet again, at least this will be a consistent behaviour.

tsusanka, 2019-10-31:
Upon discussion we have decided:

- Cancel `_boot_recovery` in `main.py` and have a single boot process
- In the main boot, if `recovery.is_in_progress() == True` then we will not set  `workflow.start_default(homescreen)`, we will do `workflow.start_default(recovery_homescreen)` instead
- `recovery_homescreen` is a replacement for the traditional homescreen/homescreen
- All messages will be registered. If Trezor receives a message during the recovery, e.g. GetAddress it will respond in the usual way (in this case something like `NotInitializedError`) and it will return to the recovery homescreen. This results in a small side effect where if the user is just entering a word, the screen will get interrupted and they have to go again. However, that applies on for messages which work on uninitialized device and show some dialog (and that should be just few of them) We believe this is acceptable.
- if a RecoveryDevice message is received while the recovery is already running, the same functions will be called as are called from the recovery homescreen but with proper wire.Context instead of wire.DummyContext

PR #662 (2019-11-07): Remove persistence boot and set recovery as a default workflow
This was a great idea! We set the recovery homescreen as a default workflow and return to that if some message is received. We do not unregister anything.

The work in workflow.py was based on a fair amount of guessing, so I would like to discuss that part mainly. I'm not sure why I can't call `close_default` (or `loop.close` respectively), in that case the emulator exits.

Closes #558.
closed 2019-11-07 by matejcik