trezor/trezor-firmware #568: Add button requests to firmware upgrade

tsusanka, 2019-09-25:
When doing a firmware upgrade bootloader does not send ButtonRequest. Wallet then does not know that the user was prompted to confirm the upgrade.

@prusnak can we add it or it is intentionally missing?

prusnak, 2020-02-04:
We are reworking the firmware chunk sizes to make the user experience better.

closed 2020-02-15 by prusnak

tsusanka, 2020-02-17:
## Testing

We have tested this with @mroz22 so this should be fine, but please test firmware upgrade just to make double sure. Let's test this on Suite@develop and Wallet@production.

Note: this requires the new bootloader to be signed so you can't test this yet. Let's test this shortly before release / at the end of freeze.

tsusanka, 2020-02-17:
@prusnak JFYI @mroz22 mentioned this is not needed on T1, that the delay there is quite low. Mybe because the flash erase is faster there?. So this was done for TT only.

prusnak, 2020-02-17:
ACK

bosomt, 2020-07-20:
@tsusanka is there still need to test it on our side 
or is it even possible ?

tsusanka, 2020-07-20:
We have not released the bootloader yet, so there is nothing test at the moment. But maybe leave the `needs QA` so we test it when we finally decide to release new bootloader.

sorooris, 2020-08-17:
any updates on this?

tsusanka, 2020-08-17:
I guess it is up to @prusnak to decide _when_ we want to release the next bootloader. But my guess is we will release when there are more changes to it.

PR #830 (2020-02-15): FW Update: Request a small chunk first to make the UI smoother
We first request 16KB which should be enough to obtain and parse the header. Then we prompt for the user confirmation and after that we request the rest. This way the Host sees the request right away after the user confirmation, meaning they know the firmware update process was confirmed by the user.

Closes #568