trezor/trezor-firmware #571: Initialize all variables explicitly

prusnak, 2019-09-25:
We need to review all C code and add explicit initialization of local variables everywhere.

We should utilize the following:

```
-Wuninitialized
-Wmaybe-uninitialized
-Wall
-Wextra
```

prusnak, 2019-09-26:
Maybe [clang-tidy](https://clang.llvm.org/extra/clang-tidy/checks/list.html) can help?

onvej-sl, 2019-09-30:
I did it partly [here](https://github.com/trezor/trezor-firmware/commits/onvej-sl/explicit-initialization).

onvej-sl, 2019-10-04:
I explicitly initialised automatic variables in `crypto`, `legacy` and `storage`. Variable length arrays (such as `char array[i];`) are initialised with `memset`. I didn't initialise static variables and variables in tests and tools.

onvej-sl, 2019-10-07:
Trezor T emulator binary increased in size by 8 kB.

prusnak, 2019-10-09:
Fixed via trezor/trezor-firmware#607

closed 2019-10-09 by prusnak