trezor/trezor-firmware #575: monero signing runs out of memory on device

prusnak, 2019-09-26:
```
Traceback (most recent call last):
  File "trezor/wire/__init__.py", line 319, in handle_session
  File "trezor/wire/__init__.py", line 427, in keychain_workflow
  File "trezor/wire/__init__.py", line 424, in keychain_workflow
  File "apps/monero/sign_tx.py", line 21, in sign_tx
  File "apps/monero/sign_tx.py", line 94, in sign_tx_dispatch
  File "apps/monero/signing/step_06_set_output.py", line 52, in set_output
  File "apps/monero/signing/step_06_set_output.py", line 264, in _range_proof
  File "apps/monero/signing/step_06_set_output.py", line 300, in _rsig_bp
  File "apps/monero/xmr/range_signatures.py", line 21, in prove_range_bp_batch
  File "apps/monero/xmr/bulletproof.py", line 1056, in prove_batch
  File "apps/monero/xmr/bulletproof.py", line 1117, in _prove_batch_main
  File "apps/monero/xmr/bulletproof.py", line 787, in vector_add
  File "apps/monero/xmr/bulletproof.py", line 616, in _ensure_dst_keyvect
  File "apps/monero/xmr/bulletproof.py", line 339, in __init__
  File "apps/monero/xmr/bulletproof.py", line 347, in _set_d
  File "apps/monero/xmr/bulletproof.py", line 347, in <listcomp>
MemoryError: memory allocation failed, allocating 2048 bytes
```

tsusanka, 2019-10-02:
We could also solve this by offloading the whole bulletproof computation to the host. If I'm not mistaken that's what we are doing at the moment if the number of inputs > 2. This has no security implications and _should_ have no privacy implications, since the amount is entered on the host anyway.

@ph4r05 might confirm this.

prusnak, 2019-10-02:
I think the choice is made in the monero client, not in the device. But I might be wrong, indeed @ph4r05 can shed some light on this.

ph4r05, 2019-10-02:
I will check it out but AFAIK we can enable full offloading on the host by modifying just the Trezor firmware by simple flag switch

ph4r05, 2019-10-02:
Pls try removing this condition: https://github.com/trezor/trezor-firmware/blob/2217b680e3e2c8131ff4b2df6963254edfe3654f/core/src/apps/monero/signing/step_01_init_transaction.py#L200

and set `state.rsig_offload = True`. It should work, offloading all Bulletproofs to the host.

ZdenekSL, 2019-10-03:
Closed in favor of 

https://github.com/trezor/trezor-firmware/issues/590

closed 2019-10-03 by ZdenekSL