trezor/trezor-firmware #579: [Feature] Stellar signing / missing operations

MisterTicot, 2019-09-30:
**Issue**

4 of the Stellar operations are not supported by Trezor: inflation, createPassiveSellOffer, manageSellOffer & manageBuyOffer.

**Notes**

- `createPassiveSellOffer` is the new name for `createPassiveOffer` - `createPassiveOffer` is still supported in Stellar official libraries for backward compatibility purpose.
- `manageSellOffer` is the new name for `manageOffer` - `manageSellOffer` is still supported in Stellar official libraries for backward compatibility purpose.
- `manageBuyOffer` is a new operation similar to `manageSellOffer` but passing orders in the other direction. Parameters are the same.

prusnak, 2019-09-30:
Please send a pull request implementing these

prusnak, 2019-09-30:
Citing from https://github.com/trezor/trezor-core/issues/202#issuecomment-393342089 why Inflation is not supported:

"Yes, any user can submit this operation to the network to trigger the inflation calculation, so there wouldn't be any point in using a Trezor to secure it. :)"

matejcik, 2019-09-30:
perhaps this should be handled in Connect -- if we're trying to match the official API. @szymonlesisz ?

MisterTicot, 2019-10-01:
I suppose that `createPassiveSellOffer` and `manageSellOffer` can be defined as aliases into Connect - but what other interfaces such as CLI then?

`inflation` is clearly a marginal operation. In fact it is in the process of being removed from stellar-core (not 100% confirmed yet) so I guess there's no point.

`manageBuyOffer` is a new operation and should probably be added to the firmware.

matejcik, 2019-10-01:
> I suppose that `createPassiveSellOffer` and `manageSellOffer` can be defined as aliases into Connect - but what other interfaces such as CLI then?

Again, our internal interfaces do not aim to match naming of the official API. The CLI tool accepts XDR data so there is no issue either.

> `manageBuyOffer` is a new operation and should probably be added to the firmware.

agreed. are you willing to make a PR for it?

MisterTicot, 2019-10-01:
Well... That was my plan before I saw all that there are a number of file to change in various languages all over the place. I'm not sure I can deal with it - in fact I'm not sure I could even list them all.

MisterTicot, 2019-10-03:
@matejcik 

Can provide me the complete list of files to edit? I'll tell you if I can deal with it.

matejcik, 2019-10-03:
> Can provide me the complete list of files to edit? I'll tell you if I can deal with it. I assume you need to edit at least both firmwares, the API and tests files right?

correct.
here's the general list of places that we would need:

- the new operation needs to be added to protobuf defs, in `common/protob/messages-stellar.proto`
- depending on the fields in the new operation, `legacy/firmware/protob/messages-stellar.options` must be adjusted: all str/bytes fields must have a set `max_size`, all `repeated` fields must have `max_count`.
- implementation must be added to `core/src/apps/stellar`
- implementation must be added to `legacy/firmware`, presumably to `*stellar*`, but i'm not sure how the implementation is structured exactly
- a test must be added to `tests/device_tests/test_msg_stellar_sign_transaction.py`, you can take inspiration in my latest PR for the other Stellar bug
- support for the new operation must be added to `python/src/trezorlib/stellar.py` XDR decoder.

this is for firmware support only. I'm not sure what needs to be done on the Connect side.

MisterTicot, 2019-10-16:
I'm going to look into this this week.

MisterTicot, 2019-10-16:
@matejcik https://github.com/trezor/trezor-firmware/pull/627

matejcik, 2021-11-04:
all operations mentioned in this issue are added in #1838

closed 2021-11-04 by matejcik