trezor/trezor-firmware #580: [BUG?] Stellar signing / set/unset signer

MisterTicot, 2019-09-30:
**Issue**

When trying to set a signer, the device doesn't display the operation but sign
the transaction. Then, the signature is invalid.

**Transaction Template**

```
{
  "source": "GAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWHF",
  "sequence": 0,
  "operations": [
    {
      "type": "setOptions",
      "signer_type": 0,
      "signer_key": "GBWYUHFOHJECKDLFNCPGPVU6XRDJIBT5BYF6VXZEDHWVQRCR4HZVCGPU",
      "signer_weight": 1
    }
  ],
  "fee": 100
}
```

**What I've tried**

- I tried weight=0 and weight=1.
- I tried with other key types (tx & hash).

matejcik, 2019-09-30:
Works fine for me both on T1 and TT on current master.

@szymonlesisz can you provide the protobufs that Connect generates for this message?

MisterTicot, 2019-10-01:
**Fail:** https://test.cosmic.link/?setOptions&network=test&signer=1:key:GAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWHF

matejcik, 2019-10-01:
Confirming the problem both on T1 and TT.

Trezor will in fact display that signing is happening, but won't show the signer key being added. Then the signature is rejected by the Stellar API.

matejcik, 2019-10-01:
The problem seems to be that Connect is not sending a value for `signer_type`. 

@MisterTicot please try again with a different key type, and the appropriate signer key. That should work AFAICT.

MisterTicot, 2019-10-01:
@matejcik 

It doesn't work here:
- https://test.cosmic.link/?setOptions&network=test&signer=1:hash:87428fc522803d31065e7bce3cf03fe475096631e5e07bbd7a0fde60c4cf25c7
- https://test.cosmic.link/?setOptions&network=test&signer=1:tx:87428fc522803d31065e7bce3cf03fe475096631e5e07bbd7a0fde60c4cf25c7

(I just notice that I inverted the type value for `tx` and `hash` signers - but in this case it doesn't change anything)

matejcik, 2019-10-01:
looking at https://github.com/trezor/connect/blob/30cb9ee389c45b1fa5eb7e01994506affe93ac3f/src/js/core/methods/helpers/stellarSignTx.js#L156 , i think you're supposed to pass the data as:
```json
"operations": [
    {
      "type": "setOptions",
      "signer": {
 	    "type": 1,
        "key": "87428fc522803d31065e7bce3cf03fe475096631e5e07bbd7a0fde60c4cf25c7",
        "weight": 1
      }
    }
  ],

MisterTicot, 2019-10-01:
That's right - I guess I mixed it up. I confirm that setting tx & hash signers works fine. Setting pubkey signer failed though.

**Input**

```
  "operations": [
    {
      "type": "setOptions",
      "signer": {
        "weight": 1,
        "type": 0,
        "key": "GBL4RZXIRGOVCA3NQIMNTPB4SJ4VUM6PUWNVFIV7U63GILTOAGAPDOWN"
      }
    }
  ]
```

**Output:**

```
Error: Illegal str: Contains non-hex characters
```

matejcik, 2019-10-01:
i think you need to send the public key as hex-encoded raw data, instead of the base32 string.

I'm closing this issue because there is no problem on the firmware side after all

> P.S.: I'm being ask to improve that flow. Please let me know if you have any clue.

Not sure what you mean. But reproducing the issues via your tool was straightforward, so :+1: 
One thing that would be helpful, the ability to edit the parameters right on the linked page.

closed 2019-10-01 by matejcik

MisterTicot, 2019-10-01:
@matejcik 

> i think you need to send the public key as hex-encoded raw data, instead of the base32 string

This is not very satisfying: in every other places pubkey are encoded as base32 string. I think this is an inconsistency that is worth a fix, isn't it?

> Not sure what you mean.

Some applications use my page to validate transactions. People complain about the flow:

My page open, then the trezor page open, then they have to validate pubkey import on the page + on the device **each time they use the page**, then they click on "sign", then they have to validate on the page again and on the device again.

Comparing to Ledger wallet flow (sorry): my page open, one enter the pin on the device, open the stellar app, the pubkey is directly imported, then click on "signing" and validate on the device.

MisterTicot, 2019-10-01:
And really, having tried each & every possible transaction on the Trezor device, I can confirm that this is a lot of steps to get a transactions signed.

matejcik, 2019-10-01:
> @matejcik
> 
> > i think you need to send the public key as hex-encoded raw data, instead of the base32 string
> 
> This is not very satisfying: in every other places pubkey are encoded as base32 string. I think this is an inconsistency that is worth a fix, isn't it?

You're sort of right, but Connect is the right place to fix that (so feel free to file an issue).

This supposed inconsistency makes sense for Trezor's protocol because the same bytes field is reused for tx, hash or publickey, and it's actually the same bytes that get written to the XDR encoding. The only difference is that public key is *displayed* in base32 while tx/hash is displayed in hex.
Think of it like this: the signer parameter always accepts raw bytes, and the signer type specifies what they mean. Unlike other places, which actually accept _addresses_.

> Some applications use my page to validate transactions. People complain about the flow:
>
> My page open, then the trezor page open, then they have to validate pubkey import on the page + on the device [...]

That is true. I'm not sure why Connect doesn't allow to "remember" the permissions, you'd have to ask elsewhere.
Also you're asking for the address with a parameter that shows it on device, which is not necessary.

MisterTicot, 2019-10-01:
Ok I got it. It all works now, thank you.