trezor/trezor-firmware #581: [BUG?] Stellar signing / manageData

MisterTicot, 2019-09-30:
**Issue**

I could not figure a way to get the right signature for setting data entries to
values that have a length that is not a multiple of 4 bytes.

In other words, when setting data entries to values whose length in bytes is a
multiple of 4, the signature is good. This including unsetting data entry (using
an empty buffer as value).

**What I tried**

- Encode value as hex.
- Write value into a 64-byte _Buffer_ then encode as hex.
- Write value into the next higher multiple of 4 byte length _Buffer_ and encode
  as hex.

matejcik, 2019-09-30:
@MisterTicot please provide an example transaction (ideally both in XDR and the JSON you are using). also please provide instructions on how to verify transaction signature :) is there an online resource we can use?

also please tell us if you're using T1 or TT

MisterTicot, 2019-10-01:
**Device:** Trezor One, latest firmware

MisterTicot, 2019-10-01:
**Fail:** https://test.cosmic.link/?manageData&network=test&name=data&value=abc
**Succeed:** https://test.cosmic.link/?manageData&network=test&name=data&value=abcd

You'll have to fund your testnet account to use this interface: https://www.stellar.org/laboratory/#account-creator

The XDR and JSON are logged in the dev console.

MisterTicot, 2019-10-01:
Oh and you'll know that the signature is invalid when the "tx_bad_auth" error is returned.

matejcik, 2019-10-01:
confirming. the same bug exists in T1 and TT, in that the value is not correctly padded to 4 bytes per XDR spec

PR #589 (2019-10-01): Stellar: fix ManageDataOp value padding 
fixes #581 

changes in core and legacy are minor, but i'd still appreciate if you took a look, esp. on the legacy code, and whether it is ok to call `stellar_hashupdate_string` like this
closed 2019-10-01 by matejcik