trezor/trezor-firmware #588: Unify setup_client behavior between T1 and TT

ciny, 2019-10-01:
```
@pytest.mark.setup_client(mnemonic=MNEMONIC12)
def test_backup_bip39(client):
    assert client.features.needs_backup is True

```
this test will pass on TT but fail on T1. We should unify this (ideally so T1 matches TT behavior). We could then drop a few skip_t1 markers and remove some duplicate tests.

matejcik, 2019-10-01:
cc @prusnak 

the problem IIUC is that for T1 `load_device` is a production call, and in that case it is not desirable to keep the device "not backed up".

but what we actually need for the backup tests is a way to initialize the device so that it can be backed up, so TT behavior is the "right" one.

Could we add a flag to `load_device` for this? Maybe that's not desirable for T1?
Otherwise we don't have the ability to load the device with a given mnemonic *and* then back it up.

OTOH maybe we don't really need that and we can just do reset-backup tests?

ZdenekSL, 2019-10-15:
Fixed by https://github.com/trezor/trezor-firmware/pull/598

matejcik, 2019-11-13:
per #598 the `setup_client` marker can now specify the state of the device with `needs_backup` or `no_backup`.

Reassigning to @ciny: please take a look and modify tests as appropriate, I don't know which ones are relevant.

ciny, 2019-11-14:
Ok, I'll have a look at it. If the defaults are False for both then only the backup tests should be affected, I'll modify/merge them as needed.

closed 2019-11-26 by ZdenekSL