trezor/trezor-firmware #590: Resolve low memory issue

prusnak, 2019-10-01:
More IRL

prusnak, 2019-10-02:
Couple of suggested fixes are in this branch:

https://github.com/trezor/trezor-firmware/tree/memory-fix

prusnak, 2019-10-03:
Handy:

`micropython.mem_info()` and `micropython.mem_info(True)`

tsusanka, 2019-10-04:
What have I encountered several times in the mypy issue is this: `content.__class__.__name__ == "Paginated"`. I'm wondering if omitting a single import has significant advantage. In other words if we need to keep doing this, because mypy does not recognize it.

prusnak, 2019-10-07:
Backporting this patch to micropython fork saved 40KB of memory at start: https://github.com/trezor/micropython/commit/64236f5518b443c3f7631a4d6b31e2c8ccc385f4 :tada:

matejcik, 2021-11-30:
This is no longer a pressing issue, and will be even less with the switch to the Rust UI.
All device tests are now passing on hardware.

There are some sub-issues: #174, #194, but both of those are find as stand-alone issues.

I'm closing this one because we don't currently need it as a separate task, AFAICT.

closed 2021-11-30 by matejcik