trezor/trezor-firmware #596: Tests: U2F's test_LongEcho sometimes fails

tsusanka, 2019-10-03:
The U2F test test_LongEcho sometimes fails. On retry it usually succeeds.

```
CHECK_LE fail at test_LongEcho[144]:received > .075: PASS(check_Compilation())
```

Can we increase the threshold or mark the test as _flaky_ to be rerun?

tsusanka, 2019-10-04:
I've added `retry` parameter in 737ebeec1c8ba91c7d5d78c598a6cabd7e4785e8 to the appropriate job in the CI as a hotfix so it should not cause problems on the CI at the moment. However, this should be fixed properly not to use the CI's power unnecessarily.

tsusanka, 2020-01-23:
This is getting more and more problematic :(. @andrewkozlik can we increase the timeout? I guess it is included in the external testing suite right?

andrewkozlik, 2020-01-23:
This looks like Google's U2F tests, which I am not familiar with. @prusnak will know more. My opinion is that it doesn't make sense to check the transfer times against the emulator in the CI. These checks should be done on a user system against a real device. So I propose adding a command-line option to HIDTest and/or U2FTest which will disable the checks in the CI.

prusnak, 2020-01-23:
I agree it does not make sense to check for timeouts in the emulated environment. Fixed that in 9a884a5dc53feb383a125bcb1d97ccb17e420f5f where I also removed retry from the CI - in case there is another HIDTest/U2FTest failing, please open a new issue.

closed 2020-01-23 by prusnak

andrewkozlik, 2020-01-23:
I don't think commenting-out the LongEcho is the proper solution. I would just suppress the last part of the test, which checks the transfer times. On the device we should run the full test including the transfer time checks.