trezor/trezor-firmware #597: Firmware fingerprint verification for Trezor One (1.8.0 and later)

serge-name, 2019-10-03:
Hello,

I have noticed that since firmware `1.8.0` the sizes of self-compiled and official firmware are different. Moreover the command `tail -c +257 trezor.signed.bin | sha256sum` gives different hashes for that firmware.

I use old repo https://github.com/trezor/trezor-mcu/ to compile version `1.8.0` and new monorepo to compile version `1.8.1` and newer.

I found that these commands work for versions `1.8.0` ~ `1.8.2`:

```shell
curl -s https://wallet.trezor.io/data/firmware/1/trezor-1.8.0.bin |tail -c +996 |sha256sum
cat /src/build/trezor-v1.8.0.bin |tail -c +740 |sha256sum
```
should match.

```shell
curl -s https://wallet.trezor.io/data/firmware/1/trezor-1.8.0.bin |tail -c +257 |head -c 544 |sha256sum
cat /src/build/trezor-v1.8.0.bin |head -c 544 |sha256sum
```
should match too.

Could you please explain why the sizes differ now? And could you please update the documentation with a new method to check firmwares?

prusnak, 2019-10-03:
New firmwares contain new firmware header, which has different size than the old firmware header.

prusnak, 2019-10-03:
Can you please send pull request to update the documentation? Thank you!

closed 2019-10-03 by prusnak

serge-name, 2019-10-03:
Ok, I will create PR soon.

> contain new firmware header

I agree, there are 256 bytes at the beginning of official firmwares that missing in self-compiled ones. Is there a firmware signature located?

And may I ask you about another thing still looking unclear?

An original firmware contains some non-zero bytes (either data or code) on the offset 800 (195 bytes total). But a self-compiled firmware contains zeroes on the according offset 544. What is the purpose of that 195 bytes?

Thanks a lot.

prusnak, 2019-10-03:
These are hashes of the firmware blocks.