trezor/trezor-firmware #599: Setting PIN on a wiped device breaks reset and recovery

andrewkozlik, 2019-10-03:
If the PIN is set on a wiped device, then:
- the reset_device message will always fail,
- the recovery_device message will fail if the pin_protection field is set to true,
- the load_device message will fail if the pin field is set.

Solutions (in my order of preference):
1. Don't allow change_pin on a wiped device.
    - This is what the code currently falsely assumes to be the case.
    - The user can still set the PIN in recovery_device, so that if they enter one or more mnemonic shares and unplug the device before completing recovery, then the entered shares will be protected by PIN. 
2. If the PIN is already set and the pin_protection field in reset_device or recovery_device is true or the pin field in load_device is set, then fail.
3. If the PIN is already set and the pin_protection field in the message is true, then prompt for the old PIN before prompting for the new PIN. If the user enters a wrong old PIN, then fail. In case of load_device always fail if the pin field is set, because there shouldn't be any user interaction.
    - Not clear why anybody would want to go through this flow.
4. If the PIN is already set, then ignore the value of the pin_protection field in reset_device and recovery_device and ignore the pin field in load_device.
    - Ignoring a field without at least warning the user seems like a bad idea. It's better to fail, i.e. Solution 2. 
5. If the PIN is set on a wiped device, then cache the PIN entered during unlock or change_pin, so that we can satisfy the request without having to reprompt the old PIN from the user.
    - Seems like a dangerous hack defeating the purpose of the PIN and definitely not worth the risk.

Before I made a full analysis, @prusnak was in favor of Solution 1 and if I recall correctly, @matejcik was in favor of allowing change_pin on a wiped device. Opinions?

tsusanka, 2019-10-04:
I'm in favour of (1) in short term since it seems like a desired behaviour. 

-----

In long term I suggest to remove all pin settings in all three messages and allow ChangePin on a wiped device. This way if anyone would want to enable PIN before running reset/recovery/load they would have to call for it explicitly. This would also simplify the code and nicely decouple these functionalities.

The obvious disadvantage is that this is a breaking change. I believe we could do this as part of the Passphrase redesign, so Suite could integrate this, however, I'm not sure how breaking we can be with regard to other third parties.

prusnak, 2019-10-04:
For legacy the simple fix would be adding `CHECK_INITIALIZED` just before this line:

https://github.com/trezor/trezor-firmware/blob/737ebeec1c8ba91c7d5d78c598a6cabd7e4785e8/legacy/firmware/fsm_msg_common.h#L139

matejcik, 2019-10-04:
I agree with @tsusanka on (a) short-term fix and (b) removing the flag from Reset and Recovery.

In LoadDevice, the field should remain AND it should override any currently set PIN. LoadDevice is a debug message whose purpose is to provision the full configuration, so it makes no sense to break the pin away.

In Reset / Recovery, I wouldn't consider it a breaking change. All software would continue to work as before, except that the "change pin" instruction would be ignored. This is undesirable in case of Shamir backup, but no software currently supports that anyway ;)
And if we wanted to keep backwards compatibility, we could use @andrewkozlik's option (3), and use the field to fire up the ChangePin flow -- which, in case pin is already set, would reprompt for old pin. Given that this is a corner-case, I'd be OK with that behavior.

andrewkozlik, 2019-10-04:
I will implement Solution 1 for now.

Note that if we will later want LoadDevice to override any currently set PIN, then the message will have to wipe the device if a PIN is set.

closed 2019-10-04 by andrewkozlik