trezor/trezor-firmware #609: Display goes black after address check with 2.1.6 fw version

bosomt, 2019-10-09:
Some users are complaining that their device screen goes black after address check for BTC and LTC coin. 
Downgrade to 2.1.5 fixes issue. Upgrade and issue is back.
I was not able to reproduce it but we have device at support with this behavior.

OS: windows and linux
Wallets; both of them
PIN enabled
Passphrase disabled
2FA enabled 
custom screen: no

See log and videos: 
https://photos.app.goo.gl/AUT8nj1mhRDLCptEA
[Jirka_log.txt](https://github.com/trezor/trezor-firmware/files/3708087/Jirka_log.txt)

prusnak, 2019-10-09:
Does the device still work afterwards? i.e. the only thing missing is the homescreen image, but other communication properly wakes up the device?

bosomt, 2019-10-09:
No it doesn't
it says Device used in another window
and after refresh of wallet it says **Your Trezor is not connected
**

prusnak, 2019-10-09:
I think this is not a firmware issue then, but a user is using two software wallets at once which clash together when they try to access the device simultaneously (maybe beta and stable wallet or our wallet + electrum?).

bosomt, 2019-10-09:
no, no other app, 
there is  one device with this behavior in office

bosomt, 2019-10-09:
stage wallet and electrum works fine.
so it's wallet related.
@TomasZorvan !

bosomt, 2019-10-09:
BUT Exodus is broken/freezing too

bosomt, 2019-10-10:
I was able to reproduce it.
As soon i used https://wallet.trezor.io or beta wallet or stage and changed homescreen.
After disabling of custom homescreen it works fine.

And yes you really need to reconnect it.
Bridge, webUSB, trezorctl sees it,  but no response from device  till it's reconnected

bosomt, 2019-10-10:
+ bitcoin only FW works fine

prusnak, 2019-10-10:
I think we run out of memory when using a custom full screen. :-/

bosomt, 2019-10-10:
it works fine in master dev FW

prusnak, 2019-10-10:
Okay, the issue is fixed then. Thanks for finding the bug and testing.

closed 2019-10-10 by prusnak

tsusanka, 2019-10-10:
And you may also inform them that it will be fixed in 2.1.7 on Nov 6th.

Perlover, 2019-10-11:
> Okay, the issue is fixed then. Thanks for finding the bug and testing.
> 
> Please tell the guy who reported it they should unset the home screen or downgrade to 2.1.5.

I have Trezor T v2.1.5 but i same bug (i have my own homescreen). But i have regulary black screen after few time after connect. But i do some commands (view address/send) - i see screen again but after command competed - black again. I see here recommendation to downgrade to 2.1.5. So do i have same bug or not?

I decided to update through `trezorctl firmware-update -f trezor-2.1.7-71086607.bin` but i see warning screen like here (https://github.com/trezor/trezor-firmware/issues/36). It's second problem. What do you recommend here?

Thanks!

prusnak, 2019-10-11:
@Perlover is the bug gone when you unset the custom homescreen?

The warning you see is legitimate - you are downloading devel version of the firmware and it will erase your seed. Don't use unless you know what you are doing.

Perlover, 2019-10-11:
> @Perlover is the bug gone when you unset the custom homescreen?

Yes, when i set homescreen from standard homescreens i don't see black screen
I have v2.1.5 version

Perlover, 2019-10-11:
> > @Perlover is the bug gone when you unset the custom homescreen?
> 
> Yes, when i set homescreen from standard homescreens i don't see black screen
> I have v2.1.5 version

And if i try to back to custom homescreen (PNG file) i got error in wallet.trezor.io: `Firmware error` (before i see in russian "Error of homescreen changing"). For information...

bosomt, 2019-10-14:
I can confirm that 2.1.5 FW version has same issue.
It doesn't goes not freeze whole device but screen is dim and doesnt show logo.
But you can wake it by sending new message

prusnak, 2019-10-14:
This will be fixed in 2.1.7 which will go live today.