trezor/trezor-firmware #610: [BUG?] Stellar signing / binary text memo

MisterTicot, 2019-10-09:
**Issue**

In js-stellar-sdk, text memo can be built either from a _String_ or a _Buffer_.
The later allow for binary values to be used. Trezor doesn't allow for a
_Buffer_ to be passed as MEMO_TEXT value, so it is unclear how this can be
achieved.

I tried to convert the _Buffer_ into ASCII characters, but the signature was
invalid.

**Transaction Template**

```
{
  "memo": {
    "type": 1,
    "text": "\u0000\u0010\u0003"
  },
  "source": "GAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAAWHF",
  "sequence": 0,
  "operations": [
    {
      "type": "setOptions"
    }
  ],
  "fee": 100
}

```

**Device Display**

`Memo (TEXT)` (Doesn't show any value)

tsusanka, 2019-10-09:
So even if the memo_type is 1 (which stands for TEXT), it is allowed to set the text via bytes?

prusnak, 2019-10-10:
I don't think you are supposed to send binary data via memo type=1. If you think otherwise please show us the relevant part of the Stellar documentation that says so.

closed 2019-10-10 by prusnak

MisterTicot, 2019-10-10:
@tsusanka  Absolutely
@prusnak Please don't close issues based on a guess lik that, this is unrespectful. I bought a Trezor device that I don't need & spend hours to make it work with Stellar. I took the time to test every case & to report the problem I had to your team. The least I expect is the answer to be respectful. Thank you.

Here's a transaction I just validated on the testnet that has base64 ABCD as text memo: https://horizon-testnet.stellar.org/transactions/b12e64bfb79794c18923f2d48dd67be31403aa4822aaa6d01b8a440bc4cf4986

You can do it as well: https://cosmic.link/setOptions&network=test&memo=base64:ABCD

Select 'Stellar Authenticator' as signing method, then 'Guest Mode', it will provide you with a funded testnet account.

MisterTicot, 2019-10-10:
I confirm that the documentation states that MEMO_TEXT should be an ASCII or UTF-8 string. However, technically, `"\u0000\u0010\u0003"` is such a string and is accepted by the network.

prusnak, 2019-10-10:
String can't start with a zero byte.

tsusanka, 2019-10-10:
@MisterTicot thanks, let's see what the Stellar devs answer.

prusnak, 2019-10-10:
> The least I expect is the answer to be respectful.

When I close the issue I am not trying to be disrespectful. If you feel that way I am sorry.

I close issues which are not issues. Issues can be reopened (if the facts change) and then we reconsider them.

MisterTicot, 2019-10-10:
> String can't start with a zero byte as it is its terminator.
> 
> Technically, you are sending an empty string, that's why it is not shown on the display.

Good catch. I confirm that I can produce valid signatures for transaction using arbitrary memos as long as they don't contains a NULL character: https://cosmic.link/?setOptions&network=test&memo=base64:TREZOR

Please note that "String can't start with a zero byte" is is a language-specific convention & limitation - not a hard rule. Many languages allow NULL in strings. See: https://en.m.wikipedia.org/wiki/Null_character.

prusnak, 2019-10-10:
Btw, what is your use case?

MisterTicot, 2019-10-10:
I provide interfaces with the Stellar Ledger (Cosmic.link, Stellar-authenticator.org, Ledger Wallet & Trezor wallet convenience libraries), so naturally I aim at a complete support of the underlying functionalities.

prusnak, 2019-10-10:
So there is no use case for this so far. I really do think MEMO TEXT field was not meant to be used with arbitrary binary strings.

MisterTicot, 2019-10-10:
> So there is no use case for this so far.

That's not what I said. I said that I don't implement features based of whether or not I can imagine an use case for it, or based on what I imagine the implementator had in mind. The network accept binary data in text memo, and the official APIs too - then my role is to indiscriminately bring this feature to the users of my service.

MisterTicot, 2019-10-10:
> fwiw, TT will sign a transaction with memo "\x00ABCD", but it will simply not show the null character.
> Not sure if it is possible to abuse this in any way, but it's likely to be the case with other string fields too - Bitcoin sign message perhaps?

Here on T1 is doesn't display anything and it signs the message, but the signature is not valid. In other cases (non-null binary characters), the display is wrong but the signature valid.

I confirm that this second case is problematic, as it is possible to cheat users about what they are really signing. If the faulty primitive is used widely, I recommend that you fix it in a timely fashion.

The best fix would be to display non-printable characters binary value in a distinguishable way - the fast fix would be to at least print a square or a warning.

MisterTicot, 2019-10-10:
Feel free to redact my previous comment once read.

prusnak, 2019-10-10:
The best fix is arguably not accepting arbitrary binary data when a string is expected. I'll open a separate issue for this.

MisterTicot, 2019-10-10:
> The best fix is arguably not accepting arbitrary binary data when a string is expected. I'll open a separate issue for this.
> 
> Thanks for the report.

That's not so easy: ASCII & UTF-8 comes with a bunch of legit yet non-printable characters that can be used to achieve the afford-mentioned attack.

prusnak, 2019-10-10:
> ASCII & UTF-8 comes with a bunch of legit yet non-printable characters that can be used to achieve the afford-mentioned attack.

We can already handle this - we print unprintable characters as `_`. Problem is just with NULL byte.

MisterTicot, 2019-10-10:
IIRC the stellar-sdk documentation used to mention `Buffer` as a possible type for MEMO_TEXT and that's why I supported it in the first place. At least, there are places in the code where this is explicitely mentioned, such as the `_validateTextValue()`: https://github.com/stellar/js-stellar-base/blob/master/src/memo.js#L126-L130.

I guess we'll see what Stellar devs answer is - but my understanding is that `HASH`, `RETURN` and `ID` types have well-defined use-cases (meaning), while `TEXT` is more a generic thing.

> We can already handle this - we print unprintable characters as _. Problem is just with the NULL byte.

I tested again | I confirm that this is working fine.

prusnak, 2019-10-10:
@matejcik let's resolve this via https://github.com/trezor/trezor-firmware/issues/614 then

MisterTicot, 2019-10-10:
This won't solve the problem that signature is invalid.

prusnak, 2019-10-10:
Okay, then we need to implement both #613 and #614

MisterTicot, 2019-10-10:
So Trezor will not support this part of the Stellar protocol?

prusnak, 2019-10-10:
It's not part of the Stellar protocol unless the specification says MEMO_TEXT should be used for passing arbitrary binary data.

MisterTicot, 2019-10-10:
https://github.com/stellar/js-stellar-base/blob/master/src/memo.js#L126-L130

prusnak, 2019-10-10:
This does not say anything about the data stored in the array/buffer.

MisterTicot, 2019-10-10:
I see

MisterTicot, 2019-10-12:
Here's the [final word](https://github.com/stellar/stellar-core/issues/2310#issuecomment-540845281): NULL character as well as any arbitrary data is permitted in MEMO_TEXT.

prusnak, 2019-10-12:
Thanks for the link, let's continue the discussion there.