trezor/trezor-firmware #613: Don't accept non-UTF-8 sequences when string is expected

prusnak, 2019-10-10:
Let's modify protobuf parsers so they will throw an error if an illegal UTF-8 sequence is passed as a string.

For T1 I opened an issue in the nanopb repo because I think this belongs in this library: https://github.com/nanopb/nanopb/issues/437
If not, we'll add the check to the application level.

@matejcik we should do this in our `protobuf.py` for TT

matejcik, 2019-10-10:
current protobuf parser will already reject invalid UTF-8 sequences

unfortunately, NULL byte is not invalid in an UTF-8 string

prusnak, 2019-10-10:
They are invalid in a zero-terminated UTF-8 string though.

prusnak, 2019-10-10:
> current protobuf parser will already reject invalid UTF-8 sequences

OKay, this is true for Model T. For T One this issue still stands.

prusnak, 2020-02-04:
Merged in 3c7806aaf6ff8482562ab8782b4ff80d5f4c9918

closed 2020-02-04 by prusnak