trezor/trezor-firmware #617: Refactor apps.wallet.sign_tx

prusnak, 2019-10-10:
Let's refactor `apps.wallet.sign_tx` into `apps.wallet.sign_bitcoin` and `apps.wallet.sign_altcoins`. This will help up with code pollution in the long run and allows us to add features likes PSBT more easily to Bitcoin.

ZdenekSL, 2019-10-22:
Related to #105

tsusanka, 2020-01-30:
I'm starting this by splitting up the sign_tx folder.

matejcik, 2020-03-20:
one random issue: when mixing segwit and non-segwit inputs, and segwit input is first, TT won't send signature for that input.

matejcik, 2020-05-05:
PR available: #980

andrewkozlik, 2020-05-05:
TODO list:
- [] Enable mypy in sign_tx code.
- [] Replace `SigningError`s and the other exception classes with `wire.Error`s directly.
- [] Modify layout functions so that they use `require_confirm` (or `require_hold_to_confirm`) in all cases, so we don't have to do `if not confirmed: raise` everywhere.
- [] Rename the app from wallet to bitcoin or bitcoinlike.
- [] Split off `apps.wallet.misc` for things that are not bitcoin-related.
- [] Add test for mixed segwit and non-segwit inputs https://github.com/trezor/trezor-firmware/issues/133

tsusanka, 2020-05-06:
I have forwarded this to QA (via the Asana form) to test Bitcoin signing including bitcoin-like coins (at least Decred, Zcash, Bitcoin Cash) on Suite, old Wallet and Electrum.

tsusanka, 2020-05-14:
> Add test for mixed segwit and non-segwit inputs #133 

@matejcik I believe this test was added in https://github.com/trezor/trezor-firmware/commit/8a0a38f948188bd250da550bb4c3d5e743b716df ? It passes on 2.3.0 and fails on 2.2.0, which seems correct. I am a bit confused about https://github.com/trezor/trezor-firmware/issues/133#issuecomment-566662032 I can't reproduce that with [your JSON](https://github.com/trezor/trezor-firmware/issues/133#issuecomment-489588086).

matejcik, 2020-05-15:
i'm not sure that it is testing all that we want, namely:
1. different orderings of segwit/nonsegwit (S->N, N->S, S->N->S, N->S->N)
2. presence of the appropriate signatures in the `signatures` part of the return value

I'm somewhat sure that this refactor fixed all these problems, but we don't know that currently. Some of these scenarios should pass on 2.2.0 as well.

tsusanka, 2020-05-17:
@matejcik I have this prepared, I just want to confirm that:

1. By S->N->S you mean three inputs, first segwit, second non-segwit, third segwit.
2. It is okay to test these against T1.
3. It is okay to use any prev_tx. In other words, the tx would fail on blockchain probably, I have made up the input details. I think we don't care about that, but I wanted to make sure.
4. We do not really care about the outputs.

matejcik, 2020-05-18:
yes to all

tsusanka, 2020-05-18:
> Enable mypy in sign_tx code.

Moved to #1001.

tsusanka, 2020-05-18:
:tada:

closed 2020-05-18 by tsusanka

tsusanka, 2020-06-02:
This has been tested as part of the _Freeze table_.

sorooris, 2020-06-03:
_*spreadsheet_