trezor/trezor-firmware #618: macOS (Catalina): error: suggest braces around initialization of subobject

brakmic, 2019-10-11:
Hi,

I've tried to compile the emulator with `pipenv run make build_unix` but it throws these error messages.

```shell
vendor/trezor-crypto/sha3.c:368:18: error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
        SHA3_CTX ctx = {0};
                        ^
                        {}
vendor/trezor-crypto/sha3.c:376:18: error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
        SHA3_CTX ctx = {0};
                        ^
                        {}
vendor/trezor-crypto/sha3.c:385:18: error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
        SHA3_CTX ctx = {0};
                        ^
                        {}
vendor/trezor-crypto/sha3.c:393:18: error: suggest braces around initialization of subobject [-Werror,-Wmissing-braces]
        SHA3_CTX ctx = {0};
                        ^
                        {}
4 errors generated.
scons: *** [build/unix/vendor/trezor-crypto/sha3.o] Error 1
```

According to [this entry from SO](https://stackoverflow.com/questions/31555584/why-is-clang-warning-suggest-braces-around-initialization-of-subobject-wmis) the initializer list should be using {{}} instead of {}.

However, after I tried to fix this particular bug it appeared in other crypto sources (blake, ed25519-donna, sha3 etc.) as well.

I'm running macOS Catalina, 10.15, with XCode 11.1 (11A1027).
My clang is: 

```shell
clang version 9.0.0 (tags/RELEASE_900/final)
Target: x86_64-apple-darwin19.0.0
Thread model: posix
InstalledDir: /usr/local/opt/llvm/bin
```

prusnak, 2019-10-11:
Can you try master now? (e688fa8) Is the issue fixed now?

brakmic, 2019-10-11:
Yes, I was able to compile the master. 
The emulator runs!

closed 2019-10-11 by prusnak