trezor/trezor-firmware #619: build-docker.sh doesn't build

Perlover, 2019-10-11:
Hello!

I tried to build the firmware of Trezor T to check hash of bin file. I used the instruction from there: https://wiki.trezor.io/Developers_guide-Deterministic_firmware_build . The `trezor-core` already archived and i tried to apply like these commands to this repo. I did: 

````
git clone --recursive https://github.com/trezor/trezor-firmware.git
cd trezor-firmware
PRODUCTION=1 ./build-docker.sh core/v2.1.6
````

And i got errors:

````
MEMORY_PROTECT=1
  CC      bootloader.o
  CC      signatures.o
  CC      usb.o
  CC      ../vendor/trezor-crypto/bignum.small.o
  CC      ../vendor/trezor-crypto/ecdsa.small.o
  CC      ../vendor/trezor-crypto/secp256k1.small.o
  CC      ../vendor/trezor-crypto/sha2.small.o
  CC      ../vendor/trezor-crypto/memzero.small.o
  CC      ../vendor/trezor-crypto/hmac_drbg.small.o
  LD      bootloader.elf
  OBJCOPY bootloader.bin
./firmware_align.py bootloader.bin
make: Leaving directory '/tmp/trezor-firmware/legacy/bootloader'
make: Entering directory '/tmp/trezor-firmware/vendor/nanopb/generator/proto'
protoc --python_out=. nanopb.proto
protoc --python_out=. plugin.proto
make: Leaving directory '/tmp/trezor-firmware/vendor/nanopb/generator/proto'
make: Entering directory '/tmp/trezor-firmware/legacy/firmware/protob'
  PROTOC  messages_pb2.py
Traceback (most recent call last):
  File "messages_map.py", line 5, in <module>
    from messages_pb2 import (
  File "/tmp/trezor-firmware/legacy/firmware/protob/messages_pb2.py", line 26, in <module>
    dependencies=[google_dot_protobuf_dot_descriptor__pb2.DESCRIPTOR,])
TypeError: __init__() got an unexpected keyword argument 'serialized_options'
make: *** [Makefile:36: messages_map.h] Error 1
make: Leaving directory '/tmp/trezor-firmware/legacy/firmware/protob'
````

Like same errors i got for core/v2.1.5 version. How can i check this repo for firmware hash?

Thanks!

prusnak, 2019-10-11:
Use `git checkout core/v2.1.6` before you run `PRODUCTION=1 ./build-docker.sh core/v2.1.6`

Reopen if this did not help you

prusnak, 2019-10-11:
Also for me the following works on the master:

```
PRODUCTION=1 ./build-docker.sh core/v2.1.6
```

closed 2019-10-11 by prusnak

Perlover, 2019-10-11:
Now the build is OK
But i don't see there any matched hash `e2032ad84108a85d4014d477b955b9181a1a56e6f222ef21bb7d47b503a02f0b` (core/v2.1.6):

````
  CC      ../vendor/nanopb/pb_decode.o
  CC      pb_encode.o
  CC      protob/messages.pb.o
  CC      protob/messages-bitcoin.pb.o
  CC      protob/messages-common.pb.o
  CC      protob/messages-crypto.pb.o
  CC      protob/messages-debug.pb.o
  CC      protob/messages-management.pb.o
  LD      trezor.elf
  OBJCOPY trezor.bin
make: Leaving directory '/tmp/trezor-firmware/legacy/firmware'
make: Entering directory '/tmp/trezor-firmware/legacy/firmware'
MEMORY_PROTECT=1
python3 ../bootloader/firmware_sign.py -f trezor.bin
Firmware size 391404 bytes
Firmware fingerprint: a30aad21897810d1717b810aae8a5d00be458b5d4fd9bba63abe7812a95c058a
Slot #1 is empty
Slot #2 is empty
Slot #3 is empty
HASHES OK
make: Leaving directory '/tmp/trezor-firmware/legacy/firmware'
````
I even did hashes for each *.bin files by this commands:

````
(base) user@host ~/tmp/trezor-firmware $ find . -name '*.bin'|awk '{ print "tail -c +1024 "$1"|sha256sum"}'|sh
e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855  -
960377ab08eac230a2782f7de3443b4a0e7f4f6714b28c4a627138aecb4d8266  -
63a4b8a4b0a9cea929695ba6e46aba7a87aa8a5632e584ea86a1cc52747fb473  -
61485a271a6f981f65273bd7638b3b903676e2ce6a2dc127c5b502d956135014  -
2c6db8b10db9fb8cc33f7c69962bc54d962049256f1bf6c7627e0fe08674adff  -
3c1b5d142fc1dff4ad2b163f79a333748e24202c0e4c1af7d0e4dfde5493643a  -
a2ceda95a4f0d06c45fefb2e8e67916e3a0d06172bd8d71b9dc2b0e7e9b51a6e  -
126784566944010213599a02a5066ae614ad7c2777800c72e423d15a80488129  -
f75da0b2843b9a846a172d3da82f8469ef2120a63c210e8528e34c2588b0d088  -
94a47df207d2612733c37358fbae5861367cf757c12315cf6a0739aad6b7a4a1  -
138d7bc9611e84211c54799b08fbf24a361e88db4f666ba6f70afd1cbf4878ba  -
(base) user@host ~/tmp/trezor-firmware $ find . -name '*.bin'|awk '{ print "tail -c +1280 "$1"|sha256sum"}'|sh
e3b0c44298fc1c149afbf4c8996fb92427ae41e4649b934ca495991b7852b855  -
a9cde5415e2ce198fae1ee9f4160d628f050679a618c14ebe8c877f63cf936ec  -
d110a6f76748f0a46d5ffee4c4f605905af9e0fa9e38f4cec72de00e22bd8709  -
6030ca6c752ec402376ca445b29690559bf687d040eeba76479f459274ce03bd  -
85a7cc41086eadde443b3a5123321a7b57e5b8508ccf9d902bcf17141c92de88  -
7ce1ba790ddd6fc7d37fdd6a109291f91995da48cefb83dfa2fe3e7f28be5375  -
012468e2aec3db77a3fb796a099d7c7ce5d5d1745b14eb793da104ed2ba904e6  -
5d89dc6f2e116dbfeac7d7e0be735a36e1e773bd1da04bc4a39d429ddddcd71a  -
538924166f93971b5fe54eff10d5e04560ed399743ad5a48239afec9cda9ad74  -
a674a6ad6e8b5f22bb433dbcc3062ed24c1ab53a65c12c76ba3210084355f4c8  -
8b96d5bb7ef7bc1ce454114333fb8df187d72eeffe88668f299148f96f44e3ab  -
````

Can you help to me what do i do wrong?

prusnak, 2019-10-11:
Where did you get this from?

`e2032ad84108a85d4014d477b955b9181a1a56e6f222ef21bb7d47b503a02f0b`

prusnak, 2019-10-11:
Look in the build log. There you'll see the hash you are looking for.

The output you are pasting 

```
python3 ../bootloader/firmware_sign.py -f trezor.bin
Firmware size 391404 bytes
Firmware fingerprint: a30aad21897810d1717b810aae8a5d00be458b5d4fd9bba63abe7812a95c058a
```

is for the T1 bitcoin-only firmware.

The script builds 4 firmwares in this order:
* core
* core-bitcoinonly
* legacy
* legacy-bitcoinonly

So after the 1st of 4 firmwares is built you'll see the hash you are looking for.

Perlover, 2019-10-11:
> Where did you get this from?
> 
> `e2032ad84108a85d4014d477b955b9181a1a56e6f222ef21bb7d47b503a02f0b`

https://beta-wallet.trezor.io/data/firmware/2/releases.json
for 2.1.6

Perlover, 2019-10-11:
> So after the 1st of 4 firmwares is built you'll see the hash you are looking for

How can i get it here?
To scroll back output logs?

prusnak, 2019-10-11:
Yes

Perlover, 2019-10-11:
I did this:

```
PRODUCTION=1 ./build-docker.sh core/v2.1.6 >& output.txt; grep fingerprint output.txt
Firmware fingerprint: 962a90386afa929d2fc9d6efbcf4c3227e7897a5b3013ec0d76745dd3e1b71fe
Firmware fingerprint: a30aad21897810d1717b810aae8a5d00be458b5d4fd9bba63abe7812a95c058a
```

But i got hashes which i cannot find here: https://beta-wallet.trezor.io/data/firmware/2/releases.json

:(

Now i have output.txt file so if you need i can send it to you

Perlover, 2019-10-11:
Here is the file output.txt

https://drive.google.com/file/d/1M-QxTtXwaS4nZOdj5fW4XQr85JTRyN0n/view?usp=sharing

Perlover, 2019-10-11:
> but in the older commit

So the `core/v2.1.6` tag in current repo doesn't match with 2.1.6 version?

Perlover, 2019-10-11:
For v2.1.5 i did:

```
git checkout core/v2.1.5
cd core/
PRODUCTION=1 ./build-docker.sh core/v2.1.5 |& tee output.txt
```

The output is:

```
arm-none-eabi-objcopy -O binary -j .flash2 build/firmware/firmware.elf build/firmware/firmware.bin.p2
cat build/firmware/firmware.bin.p1 build/firmware/firmware.bin.p2 > build/firmware/firmware.bin
tools/binctl build/firmware/firmware.bin -h
Trezor Vendor Header
  * magic   : TRZV
  * hdrlen  : 4608
  * expiry  : 0
  * version : 0.0
  * scheme  : 2 out of 3
  * trust   : 65535 = [. . . . . . . . . . . . . . . .] = []
  * vpub #1 : 47fbdc84d8abef44fe6abde8f87b6ead821b7082ec63b9f7cc33dc53bf6c708d
  * vpub #2 : 9af22a52ab47a93091403612b3d6731a2dfef8a33383048ed7556a20e8b03c81
  * vpub #3 : 2218c25f8ba70c82eba8ed6a321df209c0a7643d014f33bf9317846f62923830
  * vstr    : SatoshiLabs
  * vhash   : 0b47bb9e55673134f3d8e96bb379ec0411a4d8548feb8e52e6ffb45fdb749169
  * vimg    : (4403 bytes)
  * sigmask : 0x03 = [1 2 . . . . . .]
  * sig     : 78dd3cce48d58e880da32b279b331dbec78459520a711d5f6aef75e39626485d89480f77de74ddbc2fc2dd24e70c4d4cb8bd6d73c16d512a49f1a8d03ca5a10a
  * fngprnt : abb773af7aa4c475e505d6e1b462898bc7ebc7bc493e34d21fb2b47912d8f95b

No firmware signature
Trezor Firmware Image
  * magic   : TRZF
  * hdrlen  : 1024
  * expiry  : 0
  * codelen : 1500160
  * version : 2.1.5.0
  * fixver  : 2.1.0.0
  * hashes: OK
    - 00 : 4e70e1df196cbe01175eedf78ad3a3b480fae6b090f9ed91a305d69e6e6177bd
    - 01 : 14e6157e9833a8844e1199a77ace72b954c2917b5dd2313ef58a3b1bd95c916f
    - 02 : f3bf5bc44dbbbcd75685de0067ab8478063b0ee416774f5195f6e912ab053ddd
    - 03 : 766ae76d175650676904cbb330152b23a130ff0b7abdf4b9a0a089b9e7c66cc0
    - 04 : 6258a2a5fea63c84c3d8452ddfe7e498e709140037f7efc7aff2a19654975ec2
    - 05 : 0cc04c0ece08bf9cad4c5f08778ffbcfd055610628de546c8cfdfdfc3d067543
    - 06 : 5ea519f466aaa7b82543a42646a0bec2561cd07768e7074b972075e411d9d106
    - 07 : bb38bc7cf7979d2a6b23154c7915c829f2a3c0e1b6f73d2827b44d0a355d916a
    - 08 : 43d41c55d15bfe8381cb29277705f060b77b0ec789cbdd4452ee2c760d84d70d
    - 09 : 7c8c7279de379f9179e283638b649a06b8836bb44234c88411e6ab57dc8595ee
    - 10 : 20f2a49999ec750b40fceafa383aae6eb225f0df0af03ddcb4b43de57e9746fd
    - 11 : 17ff3f3b5986cfcf90dbe5113200ef9e1b38707cd3a802fd172402e64d915011
    - 12 : 0000000000000000000000000000000000000000000000000000000000000000
    - 13 : 0000000000000000000000000000000000000000000000000000000000000000
    - 14 : 0000000000000000000000000000000000000000000000000000000000000000
    - 15 : 0000000000000000000000000000000000000000000000000000000000000000
  * sigmask : 0x00 = [. . . . . . . .]
  * sig     : 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
  * total   : 1505792 bytes
  * fngprnt : 40e4bfaf3c5ec77872c1aaaac085aafcc443f60279ca2bb38d29c669233fdf62

dd if=build/firmware/firmware.bin of=build/firmware/firmware.bin.p1 skip=0 bs=128k count=6
6+0 records in
6+0 records out
786432 bytes (786 kB, 768 KiB) copied, 0.000572664 s, 1.4 GB/s
```

So the hash `40e4bfaf3c5ec77872c1aaaac085aafcc443f60279ca2bb38d29c669233fdf62` is in the https://beta-wallet.trezor.io/data/firmware/2/releases.json . It's fine - the proof! ;-)

But for v2.1.6 it doesn't work :(

prusnak, 2019-10-11:
This works for me:

```
git checkout core/v2.1.6
git submodule update --init --recursive
PRODUCTION=1 ./build-docker.sh core/v2.1.6 |& tee output.txt
```

output:

```
Trezor Firmware Image
  * magic   : TRZF
  * hdrlen  : 1024
  * expiry  : 0
  * codelen : 1542144
  * version : 2.1.6.0
  * fixver  : 2.1.0.0
  * hashes: OK
    - 00 : 22f690ae432a5cbbbc263dc1ed8db36f6faa994b91df850eaa43c0ef230063ed
    - 01 : a086da10a4fd3bd2255ba51136b0b0c872ba450125714963547c87e7882d7577
    - 02 : 9ee4b01d5343c1a098d4a45e0ea2a43366d7af574fe79e903540f005bfae887f
    - 03 : 42b3b0c718e0bb03c805340f9fbf812f8f2d97fcf1823a576c4bc20482325837
    - 04 : 287724a770ac52e443c67e8958a93945303dad8ae909300c34253cae02e6a32a
    - 05 : 73e51c3c36bc0c17d5822173a40ffe171471662c7a7858c5bf21fdb128a6e1d3
    - 06 : 9b9a34fb19ac35aebc66e01fa81b01923ff2f4c14522597b2bd2f7d910f52de1
    - 07 : 09c2089cfc3fa5caecd25c2a936886dd824443364c15ed200588d8261050219a
    - 08 : 4f0ff21c86ae0e1b199e6cba2983b80453315735eb89a6498b14d6987f782389
    - 09 : c4b12d1b03ccc99d504dfdcdc043f381c298b2c510c7a0760bcdd4d8fdb26a2e
    - 10 : d202b83df55f3d0046d9612dfd1a22c3beac83161627311268f60888c8cb83d3
    - 11 : 76add3be693121f065908ee76a4d0d8a4c61476366683bf44c80dcbedfc18fcf
    - 12 : 0000000000000000000000000000000000000000000000000000000000000000
    - 13 : 0000000000000000000000000000000000000000000000000000000000000000
    - 14 : 0000000000000000000000000000000000000000000000000000000000000000
    - 15 : 0000000000000000000000000000000000000000000000000000000000000000
  * sigmask : 0x00 = [. . . . . . . .]
  * sig     : 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
  * total   : 1547776 bytes
  * fngprnt : e2032ad84108a85d4014d477b955b9181a1a56e6f222ef21bb7d47b503a02f0b
```

which is the hash in the releases json

My guess is you did not run `git submodule update --init --recursive` after you checked out the `core/v2.1.6` tag.

Perlover, 2019-10-11:
Yes, i think now too!
I will check it tomorrow.

Perlover, 2019-10-14:
Yes, i did `git submodule update --init --recursive` and no i have same hash too!

```
No firmware signature
Trezor Firmware Image
  * magic   : TRZF
  * hdrlen  : 1024
  * expiry  : 0
  * codelen : 1542144
  * version : 2.1.6.0
  * fixver  : 2.1.0.0
  * hashes: OK
    - 00 : 22f690ae432a5cbbbc263dc1ed8db36f6faa994b91df850eaa43c0ef230063ed
    - 01 : a086da10a4fd3bd2255ba51136b0b0c872ba450125714963547c87e7882d7577
    - 02 : 9ee4b01d5343c1a098d4a45e0ea2a43366d7af574fe79e903540f005bfae887f
    - 03 : 42b3b0c718e0bb03c805340f9fbf812f8f2d97fcf1823a576c4bc20482325837
    - 04 : 287724a770ac52e443c67e8958a93945303dad8ae909300c34253cae02e6a32a
    - 05 : 73e51c3c36bc0c17d5822173a40ffe171471662c7a7858c5bf21fdb128a6e1d3
    - 06 : 9b9a34fb19ac35aebc66e01fa81b01923ff2f4c14522597b2bd2f7d910f52de1
    - 07 : 09c2089cfc3fa5caecd25c2a936886dd824443364c15ed200588d8261050219a
    - 08 : 4f0ff21c86ae0e1b199e6cba2983b80453315735eb89a6498b14d6987f782389
    - 09 : c4b12d1b03ccc99d504dfdcdc043f381c298b2c510c7a0760bcdd4d8fdb26a2e
    - 10 : d202b83df55f3d0046d9612dfd1a22c3beac83161627311268f60888c8cb83d3
    - 11 : 76add3be693121f065908ee76a4d0d8a4c61476366683bf44c80dcbedfc18fcf
    - 12 : 0000000000000000000000000000000000000000000000000000000000000000
    - 13 : 0000000000000000000000000000000000000000000000000000000000000000
    - 14 : 0000000000000000000000000000000000000000000000000000000000000000
    - 15 : 0000000000000000000000000000000000000000000000000000000000000000
  * sigmask : 0x00 = [. . . . . . . .]
  * sig     : 00000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000000
  * total   : 1547776 bytes
  * fngprnt : e2032ad84108a85d4014d477b955b9181a1a56e6f222ef21bb7d47b503a02f0b
```

prusnak, 2019-10-14:
Great!