trezor/trezor-firmware #629: test FD

sl-freshdesk-integration, 2019-10-16:
<div> Freshdesk Ticket ID: <a href="https://satoshilabs.freshdesk.com/a/tickets/4762">4762</a></div><div> Freshdesk Ticket Agent: Jan</div><div> Freshdesk Ticket Agent Email: jan.kovarik@satoshilabs.com</div><div> Ticket Priority: Low</div><div> Freshdesk Ticket Description: </div><div>Test</div>

sl-freshdesk-integration, 2019-10-16:
<div>Agent Jan linked Freshdesk ticket <a href="https://satoshilabs.freshdesk.com/a/tickets/4762">4762 </a>for this issue</div>

sl-freshdesk-integration, 2019-10-16:
<div>Agent Jan linked Freshdesk ticket <a href="https://satoshilabs.freshdesk.com/a/tickets/7280">7280 </a>for this issue</div>

yohaaan, 2019-10-16:
test - pls close

closed 2019-10-16 by tsusanka

sl-freshdesk-integration, 2020-02-17:
<div>Agent Jan created comment </div></br><div style="font-family:-apple-system, BlinkMacSystemFont, Segoe UI, Roboto, Helvetica Neue, Arial, sans-serif; font-size:14px"><div dir="ltr">dsfsfsdfsdfs</div></div>