trezor/trezor-firmware #630: Invalid firmware hash

AndreAhmed, 2019-10-16:
Hello,
I'm trying to compile trezor for F4 and I get in the bootloader invalid firmware hash. I debugged it, and I found block number 6 has invalid hash.

What are the steps for building the bootloader,boardloader, firmware so that I can get a correct hashing and signing of the firmware

prusnak, 2019-10-16:
`make build_boardloader build_bootloader build_firmware`

closed 2019-10-16 by prusnak

AndreAhmed, 2019-10-16:
I did the same and I still get invalid firmware hash. What could be the issues from your experience.

prusnak, 2019-10-16:
Please post the full log: what exact commands are you running and what you see as the output.

AndreAhmed, 2019-10-16:
I run the above commands that you stated exactly.
The boardloader checks the bootloader integrity correctly, but the bootloader fails to find the integrity of the firmware. ensure(check_image_contents) with firmware address and sectors, fails, and it says invalid firmware hash at block number 6

prusnak, 2019-10-16:
Sorry, I can't help you. For me it works. Please let us know once you find the problem/