trezor/trezor-firmware #631: IDEA: Create testing "suite"

tsusanka, 2019-10-17:
This was brought up by @szymonlesisz and we probably should not call this "suite".

----

The goal is to provide an interface for Connect or any other platform to run different emulator versions both T1 and TT and send debug commands.

The idea is to either expand _python_ or create a new project which would spin up a websocket and wait for commands. That way we could have integration tests for Suite which would work against an emulator and test the workflow completely. An example usage is we spin up emulator in version 2.0.0 and validate that Suite offers update.

This could be also useful for external developers - we may instruct them to simply use this tool to spin up version X. We may use firmware.corp.sldev.cz to store the emulators (but it's behind VPN, so TBD).

----

A MVP would:
- wait on websocket for commands
- launch emulator T1/TT
- launch different versions
- send commands on debug link to Trezor

In future we may add:
- launch master
- start bridge on specific version
- load device with specific firmware

matejcik, 2019-10-17:
dup of https://github.com/trezor/trezor-firmware/issues/115

tsusanka, 2019-10-17:
My bad :(.

closed 2019-10-17 by tsusanka